MarginallyClever/Makelangelo-software #67: Clearly Differentiate Makelangelo's Hardware and Software Version Scheme

virtuoushub, 2015-04-07:
There is a [4.0 Milestone](https://github.com/MarginallyClever/Makelangelo/milestones/4.0), whose versions, if I am not mistaken, reflects that of the hardware kit to make a Makelangelo. 

In the milestone are listed some tasks which are specific to the Java software used to control the Makelangelo hardware. The currently released version of this software is v6 where the development branch is v7-SNAPSHOT.

I'd like a way to better distinguish between the hardware and the software version, as I am currently contributing more to the software side. Potentially a v7.0.0, v7.0.1, v7.1.0, v8.0.0, ... of the software could be released between now and the current milestone date of 2015-05-14. 

This issue seems to be touched upon in the "move to better maven milestone system" bullet and we [previously discussed](https://github.com/MarginallyClever/Makelangelo/pull/63#issuecomment-86090414) some of my thoughts on version naming schemes.

# Milestone details taken from [here](https://github.com/MarginallyClever/Makelangelo/milestones)
* Hardware Specific
    * remove all glue and zip ties from assembly
    * remove any need to crimp, solder, strip, or cut anything
    * improve drawing on larger vertical surfaces (windows, whiteboards)*
* Software Specific
    * add check-for-update and auto-update
    * move to better maven milestone system
    * move robot-specific controls to context-sensitive menu

*Could this potentially be an overlapping task where certain software configurations aid in the drawing on larger vertical surfaces?

i-make-robots, 2015-04-11:
What do you propose?

virtuoushub, 2015-04-12:
@i-make-robots, I am not sure.

One idea I had was splitting the respective parts into their own projects. So, depending on how granular you wanted to be, you could do something like:
***
### Project Breakdown
* Hardware (e.g. Version 4.0)
    * This project would consist of the Arduino/RUMBA/etc firmware as well as any artifacts like CAD files for the hardware kit.
* GUI (e.g. v7)
    * This project would just contain the Java application with a graphical user interface for connecting/communicating with the Makelanglo hardware. 

***
or
***

### Project Breakdown ( more granular )
* Firmware (e.g. Version 4.0)
    * This project would consist of just the Arduino/RUMBA firmware.
* Hardware Artifacts (e.g. Version 4.0)
   * This project would consist of just the artifacts like CAD files for the kit.
* GUI (e.g. v7)
    * This project would just contain the Java application with a graphical user interface for connecting/communicating to the Makelanglo hardware.

***

i-make-robots, 2015-05-02:
Changed github milestones to match software version.

virtuoushub, 2015-05-12:
per @i-make-robots [comment](https://github.com/MarginallyClever/Makelangelo/issues/67#issuecomment-98387009) I am closing this issue.

closed 2015-05-12 by virtuoushub