MarginallyClever/Makelangelo-software #70: java.io.BufferedReader is never closed.

virtuoushub, 2015-04-19:
A resource leak may occur if you do not close your IO streams.

When reading in the temporary G-code file in [Filters.Filter#TextDrawLine(String, OutputStreamWriter)](https://github.com/MarginallyClever/Makelangelo/blob/dev/java/src/Filters/Filter.java#L488) the [java.io.BufferedReader](http://docs.oracle.com/javase/7/docs/api/java/io/BufferedReader.html) is never closed.

##### Offending line: Filters.Filter:555
[```BufferedReader in = new BufferedReader(new InputStreamReader(new FileInputStream(fn),"UTF-8"));```](
https://github.com/MarginallyClever/Makelangelo/blob/dev/java/src/Filters/Filter.java#L555)

i-make-robots, 2015-04-19:
I saw this warning in eclipse.  Every time I fixed it eclipse also borked one unicode character elsewhere in the file.  Did it try to do that for you, too?

virtuoushub, 2015-04-20:
So yes and no. I had to do some janky manually editting of a patch file to remove the auto convertion of the [ellipsis character](http://en.wikipedia.org/wiki/Ellipsis). The offending line is [Filter:543](https://github.com/MarginallyClever/Makelangelo/blob/master/java/src/Filters/Filter.java#L543).

FWIW I believe the character is not actually Unicode in the source file. If I look on Github it looks to be: 
```java
case 'É':  name="SPACE";  break;
```
when really I believe it should look like: 
```java
case '…':  name="SPACE";  break;
```

PR #71 (2015-04-19): Close input stream using try-with-resources statement.
Closes #70
closed 2015-04-20 by virtuoushub