MarginallyClever/Makelangelo-software #74: reset all preferences option

i-make-robots, 2015-04-20:
We miss test cases.  An automated test system would want to be able to reset all preferences and delete all temporary files so that it could test clean.

Could this be done as a deeply hidden menu option or would it need to be a separate app?  I'm concerned some classes would finalize() and safe prefs() after whatever bombedBackToTheRoaches() method is called.

virtuoushub, 2015-04-20:
@i-make-robots I am in the process of coming up with some unit tests. I plan to try and pick certain [single responsibility principled methods](http://en.wikipedia.org/wiki/Single_responsibility_principle) to get started with. That and I also need to hook up a running instance of the application to a profiler and see which pieces of code get called the most. Hello [Pareto principle](http://en.wikipedia.org/wiki/Pareto_principle). :smile: 

When looking at the code using java.util.prefs.Preferences#userRoot in relation to #72 I noticed some places that could use a little love. However without unit tests for regression I did not feel comfortable performing surgery.

I will report back when I have some unit tests completed.

virtuoushub, 2015-06-17:
See:
* [com.marginallyclever.makelangelo.PreferencesHelperTest#clearAll(Preferences)](https://github.com/MarginallyClever/Makelangelo/blob/dev/java/src/test/java/com/marginallyclever/makelangelo/PreferencesHelperTest.java#L163-#L185)
* [com.marginallyclever.makelangelo.PreferencesHelperTest#shallowClearPreferences(Preferences)](https://github.com/MarginallyClever/Makelangelo/blob/dev/java/src/test/java/com/marginallyclever/makelangelo/PreferencesHelperTest.java#L187-#L199)
* [com.marginallyclever.makelangelo.PreferencesHelperTest#deepClearPreferences(Preferences)](https://github.com/MarginallyClever/Makelangelo/blob/dev/java/src/test/java/com/marginallyclever/makelangelo/PreferencesHelperTest.java#L201-L217)

None of those methods are currently being called as they are destructive and I have not publicly implemented the ["export preference nodes to flat file"](https://github.com/MarginallyClever/Makelangelo/pull/115#issuecomment-112886689) method to backup preference node values.

i-make-robots, 2015-06-17:
I'll wait.  I'm still using current prefs for the public displays I'm doing
in downtown St. Petersburg.  Playing it safe.

You are some kind of genius, you know that?  I have no talent for what you
do.  Blows my mind every time.

Dan

i-make-robots, 2015-07-21:
Any news on this save to flat file (json?) and back?  I'm blowing up my
settings during my downtime because I've got a machine numbered -1 and it's
breaking the production build.  No option to delete a single machine number
from configs.  Whoops.

virtuoushub, 2015-07-23:
@i-make-robots see #127 to store preferences as flat file and #125 on how to remove machine's numbered anything less than 0.

i-make-robots, 2015-09-23:
Save to prefs and clear prefs work.  Does reload prefs work for you?  We may be able to close this issue.

i-make-robots, 2015-10-17:
Closing because my tests say it's ok.  Will reopen if needed.

PR #114 (2015-06-16): dev v7.1.4-SNAPSHOT
### Changelist:
* Closes #74 
* Minor bug fixes
* Code house keeping

# ~~SIMILAR WARNING AS #108~~

~~To be clear, this PR removes the following files:~~

I left some of the following documentation to capture how to bring back Eclipse specific files.

8ad26076cd7fd09f497bbe0046533a2ba06c49a5 removed the following files:
```
java/.project
java/.settings/*
```

~~I implore you to resolve #111 in one way or the other. My vote is no IDE/developer specific files in the repo.~~

~~If you accept this pull request make sure to have Eclipse closed before you update your Git repo. A [... cut]
PR #132 (2015-08-30): the art vandelay update
import, export, and resetting of preferences can be done through the GUI:
* `Makelangelo` drop down file menu
    * `Preferences`
        * ...
        * `Import Preferences`
        * `Export Preferences`
        * `Reset Preferences`

[image]

PS added some fault tolerance so that if no language folder is found it loads a default english translation contained in the jar.

PPS the GUI is not immediately responsive to the fact that preferences have been reset/imported. Work needs to be done so that updates are reflected.

Closes #74
closed 2015-10-17 by i-make-robots