MarginallyClever/Makelangelo-software #79: Wiki update

virtuoushub, 2015-04-24:
Some of the documentation on the wiki is now outdated. Some slight tweaks are necessary to have it be pertinent to the current codebase. 

See:
* https://github.com/MarginallyClever/Makelangelo/wiki#download
* https://github.com/MarginallyClever/Makelangelo/wiki#steps

i-make-robots, 2015-05-02:
Good point.  Can you please suggest changes?

virtuoushub, 2015-05-10:
I updated the wiki with some of what I was thinking about as being necessary for getting up to speed as a developer. I believe the tweaks you and I made update the wiki sufficiently for the time being. I am closing this for now, however if the wiki needs to be updated and revisited in the future I will link back to this issue and further flesh out some suggested changes.

closed 2015-05-10 by virtuoushub

i-make-robots, 2015-05-10:
maybe a note in pom.xml to say "hey!  You're changing maven, maybe update
the wiki, too."