MarginallyClever/Makelangelo-software #80: Provide user with better notification when they are converting an image file into G-code

virtuoushub, 2015-04-24:
I have noticed that after pressing "Load File..." under the GCODE tab, and selecting you conversation style/configuration, the application will no indicate to the user that anything is happening unless you explicitly click on the Log tab.

I found it not to be as intuitive as I would want. One possible solution is to open the log after you open a file.

i-make-robots, 2015-04-24:
A progress bar would also be good.  there are many non-translated strings in here. :T

i-make-robots, 2015-05-02:
I will auto-tab to Log window when a conversion begins.  A progress bar with a cancel button would be better, but AFAIK it means putting all conversions into a separate thread and that's a lot of work that won't happen before mile 8

closed 2015-05-10 by i-make-robots