MarginallyClever/Makelangelo-software #84: add auto-update

i-make-robots, 2015-05-02:
check-for-update already exists.  it's very crude, and should rely on the github repo rather than pinging my company's web server.

virtuoushub, 2015-05-09:
#### Possible Options:
* [Getdown](https://github.com/threerings/getdown)
* Java Web Start

devburke, 2016-05-05:
The check for update function never works for me, even when I know there's an updated release on GitHub. Maybe related, but the software always identifies itself as version 1.0.

i-make-robots, 2016-05-05:
What version of the software are you using now?

Can you run the app from the command line and see if error messages
appear?  (Sadly, they are not dumped to a log file or emailed to me.  YET.)

devburke, 2016-05-05:
I thought it was current, must've been a version behind. I just manually updated, and it seems to identify the version correctly now.

Although now it always opens in a really goofy sized window: http://i.imgur.com/DwJyO44.png
I've run the latest version on Windows at work and I get weirdly sized windows there now too, although not in the same skinny shape. It doesn't seem to remember anything about how it was previously opened, it always opens weird. I guess this should be a separate ticket though.

devburke, 2016-05-05:
Oh wait now I see. I was using OS X's "About" menu, not the one inside the app's own Java toolbar. I guess I can't vouch for auto-update working after all then.

i-make-robots, 2016-05-05:
The latest should
A) not make a window that goes off the screen and
B) remember your window position and size from the last time you ran the app

You're saying it restores incorrectly? Boo!

i-make-robots, 2016-05-05:
I don't know how to make the main menu appear in the OS X toolbar :(

devburke, 2016-05-06:
Yep. It launches kinda goofy on the first run, I resize it, and it keeps launching the same kinda goofy on every relaunch.

I’m afraid this is beyond my current knowledge as far as coding too. I never got to a GUI level in my programming knowledge.

devburke, 2016-05-06:
No worries! I just wanted to make sure I was reporting consistent information.

i-make-robots, 2016-05-06:
I wonder if your computer is preventing preferences from being written.  It
might be failing in a strange way and then trying to create a window that's
0x0.

devburke, 2016-05-07:
I thought maybe it was somehow happening cuz I’d renamed the jar to simply “Makelangelo”, but I tried it without renaming just now and got the same result. Nothing came up in the Console either. There were similar results on both OS X and Windows, although the way the window comes up weird is different on each, but in both cases, it comes up weird and doesn’t remember how I set it.

i-make-robots, 2016-05-07:
I've tried to tweak it in the dev build.  It passes every test here, so
either
(a) we wait and see what the next release does for you
or
(b) you build the dev branch and try it for yourself right now

Instructions on how to build it are located in the github makelangelo wiki.
Then when I push changes to the code you could pull them, try it out, and
let me know if it's OK now.  Much faster iteration time.

devburke, 2016-05-12:
Sorry, I just haven’t had time to look at this lately, I’ve been busy with other projects. I just started running ours here for the first time in awhile after updating the software, and noticed that it seems to have changed the logic of how it draws, and doesn’t work as efficiently. I’ll hake a separate ticket for that.

i-make-robots, 2016-05-18:
Confirming there is an update available is as far as I want to go with this.  I need all the exposure I can get, and driving people to my site is all I've got right now.

closed 2016-05-18 by i-make-robots

devburke, 2016-05-19:
At the moment, it doesn't detect that there's an update available. I've run older versions and checked for updates, and it always tells me I'm running the most current version.

i-make-robots, 2016-08-04:
From #196, http://www.dreamincode.net/forums/topic/190944-creating-an-updater-in-java/

I have written this, but it will sit on my HD.  I need to drive non-paying users to my website so that (maybe) I can convert them to paying customers.  Such are the challenges of being a business owner.