MarginallyClever/Makelangelo-software #89: "Adjust machine and pulley size" won't work in 7.1.1

pkmcnc, 2015-05-09:
Just throws a bunch of errors to command window.

"About" window looks weird http://i.imgur.com/rPgM6Oq.png

And I can't see recently used files list...? So pity if it's gone.

virtuoushub, 2015-05-09:
The "About" window looking weird is due to the translation files included in the release zip not containing [these lines](https://github.com/MarginallyClever/Makelangelo/blob/master/java/languages/english.xml#L628-666). A workaround is to replace the ones included in the zip with the ones on [GitHub](https://github.com/MarginallyClever/Makelangelo/tree/master/java/languages).

I have not investigated what is going on with "Adjust machine and pulley size" yet.

i-make-robots, 2015-05-09:
I'll get started on these on monday.  I need the weekend to deal with the
emotional BS of putting out a disappointing build.  Not sure what to do
about these bugs.

virtuoushub, 2015-05-10:
@i-make-robots I fixed the bugs in my branch. See the Pull Request I just opened: #91

The only thing needed before merging that into the master branch is testing that my fixes do indeed resolve the bugs, and then tagging/packaging the jar as version 7.1.2 instead of the currently packaged 7.1.2-SNAPSHOT.

I am available for assistance with packaging. Depending on your build server's speed; it takes less than a minute to complete with a couple more minutes for committing/tagging the update appropriately. I estimate between 10-15 minutes tops.

PR #91 (2015-05-10): v7.1.2-SNAPSHOT (The Exterminator)
##### Change Notes
* Updated release and source translation files to reflect latest from dev branch and make sure release/source files are in sync.
* Updated release binary to v7.1.2-SNAPSHOT version built off of my fork's commit [3023f1aa68114e650af56ed2eaccf674a70ec90e](https://github.com/MakeHartford/Makelangelo/tree/3023f1aa68114e650af56ed2eaccf674a70ec90e).*
* Minor housekeeping.
* Closes #89, #90 

*If you test this snapshot jar and can verify that it closes #89, and #90 I will open another PR with a release version of v7.1.2 so that you have an officia [... cut]
closed 2015-05-29 by i-make-robots