MarginallyClever/Makelangelo-software #92: load gcode with no machine configurations causes exception

i-make-robots, 2015-05-15:


virtuoushub, 2015-06-22:
@i-make-robots, does this still happen in the development branch?

i-make-robots, 2015-06-23:
I don't know, I'd have to wipe out all my configs to test it.

virtuoushub, 2015-09-01:
Have you tested this now that you can import/export configurations via #132?

i-make-robots, 2015-09-01:
I've exported and wiped. on wipe the menus don't update.  Maybe it should
popup a "restart now?" option to reset the menus.

Can we make export and import into "save" and "load"?  We already have
translations for those strings :)

i-make-robots, 2015-09-01:
Seems to work great.  Nicely done!

i-make-robots, 2015-09-23:
Is there an automated test?

i-make-robots, 2015-12-23:
Best solution is to have a default config already available on first program start.  Then there will always be a MachineConfiguration loaded.

i-make-robots, 2016-06-21:
will reopen if the bug reappears, but we haven't seen or heard of it in a long time.

PR #108 (2015-05-30): Feature/first start fixes
Added/fixed logic for on first load of application correctly prompt for language and do not crash when there is no known machine configurations.

Fixes #92
Includes registry workaround for #93
Ensures #105 is not reopened.

I also did some house cleaning and removed/ignored developer specific Eclipse IDE configuration files: 

* java/.project
* java/.settings/*

# WARNING
If you accept this pull request make sure to have Eclipse closed before you update your Git repo. After you the merge is successful and you update your local repo, you can checkout the four files in  [... cut]
closed 2016-06-21 by i-make-robots