MarginallyClever/Makelangelo-software #93: first load on new install causes non-fatal exception

i-make-robots, 2015-05-15:
can't create prefs root node.  I don't have the exact text.  windows 8.1 w/ java 1.8

virtuoushub, 2015-05-19:
possibly related to #72.

Closing #74 goes a long way to ensure future releases do not contain preference node related issues. When you get a chance please look into accepting #95 as I think it will help when creating unit tests for these types of things.

i-make-robots, 2015-05-19:
I would have gone faster, I was travelling all day yesterday.

virtuoushub, 2015-05-25:
Unless this is different that what @chrsmlls333 [observed](https://github.com/MarginallyClever/Makelangelo/issues/97#issuecomment-104853682), than this is [not actually an issue](https://github.com/MarginallyClever/Makelangelo/issues/97#issuecomment-105095394).

Also this is most likely not related to #72.

i-make-robots, 2015-05-26:
expected or not, we'll keep hearing about it until it stops happening.  Is there a known fix?  asking everyone to modify their java settings is unreasonable.

virtuoushub, 2015-05-30:
One suggested fix is creating the missing registry key. On Windows, which to my knowledge is the only place this warning occurs, you need to do this as an administrator. IMHO this is just as unreasonable as asking everyone to modify the logging settings for their java.

Another is to change the logging level for the `java.util.prefs`, I am not sure if this can be done outside of the JVM level however I have created a [PR](https://github.com/MarginallyClever/Makelangelo/pull/107) to see if our logging framework facade, SLF4J, will handle it.

To be clear the message @chrsmlls333 was seeing was only a warning and had no effect on Makelangelo's behavior. I agree that making it stop happening is ideal, right now I have tried to document and workaround the issue. I do not really like changing the logging level for a specific package just to make the warning not visible however I see no alternative option short of having a Windows administrator create the missing registry keys.

i-make-robots, 2015-05-30:
Could the windows start.bat file do it?

virtuoushub, 2015-05-30:
Yes, the logic to add the missing registry key could be added to [start windows.bat](https://github.com/MarginallyClever/Makelangelo/blob/master/last_official_release/Makelangelo/start%20windows.bat), however in order for it to work the batch file needs to be run by an administrator. This might not always be possible. I do not recommend this solution.

Thinking about it more, the extended Preferences API idea seems like the way to do it. While writing unit tests to help close #74, an API could also be written.

I was thinking files containing the preferences as JSON data would be used. Do you have any objection to this approach? In this way preferences like machine configurations could be easily imported/exported and viewed/edited in a pretty human readable format.

i-make-robots, 2015-05-31:
I don't mind.  The same JSON could later be sync'd in the cloud for those
who want to back up their settings.

i-make-robots, 2015-09-09:
I have tested for this bug and am unable to reproduce it any more.  Is it closed?  Does clearing out the prefs actually delete the root pref node for makelangelo and recreate this situation?

i-make-robots, 2015-09-23:
Tested, seems OK.  Is there an automated test to keep this bug dead?

closed 2016-03-30 by i-make-robots