MarginallyClever/Makelangelo-software #97: Disengage Motors on Disconnect

chrsmlls333, 2015-05-20:
The stepper motors are engaged on first movement and then never disengaged if disconnected.

Not a critical issue but it might also be useful to add engage/disengage buttons for manual readjustment purposes.

i-make-robots, 2015-05-21:
Send M18 to deactivate steppers.  Send M17 to reactivate.  
Both could be made into buttons on the Drive panel.  Translations would also need to be added.
Can you add these and make a pull request?

chrsmlls333, 2015-05-22:
I gave it a shot but, as a novice, I'm not quite sure how to build it to test. I use Maven Install in the Run As menu but all I get is "...7.1.3-Snapshot..." jars in the /target folder that don't have my changes. If you or @virtuoushub have posted any info on this please let me know. Otherwise no rush and I'll keep fiddling.

i-make-robots, 2015-05-22:
Thanks you Chris for making an attempt at this.  @virtuoushub and I have
only just added Maven and your first-time-use feedback is really helpful.
Can you give any more detail about what happened?  Can you run in debug
after maven > update project?

chrsmlls333, 2015-05-23:
Ok so I had tried that multiple times before but I think this is what solved it: 
- I ran Clean,
- I opened Build Path and made the source folder not "/src" but "/src/main/java" so all the import statements worked out (I hadn't done this before)
- I ran Maven > Update and now all import warnings are gone but Build Path refers to JavaSE rather than the JDK
- Correct that and run Debug as Java Application: com.marginallyclever.makelangelo.

Previously it would say it couldn't find main, but step 2 seemed to solve that.

It runs great except for issue #93, which gives me:
```
May 22, 2015 11:17:51 PM java.util.prefs.WindowsPreferences <init>
WARNING: Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5.
```
(on Windows 8.1, JDK1.8.0_25)

So I can test now, which is great and I'll get on that. How are you exporting your builds?

chrsmlls333, 2015-05-23:
Also: any thoughts on adding M17 to the commands sent on first handshake to properly "activate" the machine? Not really necessary but an idea. And/or M18 on disconnect?

i-make-robots, 2015-05-23:
I'd leave them optional. If you force them to happen someone's going to yell "Put it back the way it was!"

i-make-robots, 2015-05-23:
We only just moved to maven, so there are still some rough parts. The arc folder appears to be one. Exporting the project seems to be another. @virtuoushub is the maven expert. I cc'd him, I hope he can answer your question.

virtuoushub, 2015-05-25:
@chrsmlls333, @i-make-robots, thank you for including me on this issue. I will get some more thorough information on Maven setup/jar export put up on the wiki.

In regards to the [Windows registry warning](https://github.com/MarginallyClever/Makelangelo/issues/97#issuecomment-104853682), it is actually not an issue. I did a little [digging](http://stackoverflow.com/questions/18559000/reading-and-writing-from-to-java-preferences-fails-on-fresh-windows-8/18834971#18834971), and it is [expected behavior](http://bugs.java.com/bugdatabase/view_bug.do?bug_id=6809488). What I can do is document this warning as being known and include a workaround that consists of a [.REG file](http://en.wikipedia.org/wiki/Windows_Registry#.REG_files) which imports the missing key that causes the warning. FWIW, the registry key import needs to be run by an administrator.

i-make-robots, 2015-06-25:
Arduino has no reliable way to detect disconnect.  Best we can do is disable them on connect, which creates a number of edge cases and means adding new logic to enable motors before every move command, which makes a mess.

long story short, I'm not inclined to fix it.

chrsmlls333, 2015-06-27:
@i-make-robots That's completely understandable. Yeah enabling on every move would be unnecessarily complicated. Feel free to close this issue then.

closed 2015-06-29 by i-make-robots