MarginallyClever/Makelangelo-software #105: When loading an image file to convert to G-Code, Tools are being listed as machine configurations.

virtuoushub, 2015-05-25:
[image]

#95 is most likely to blame.

Possibly related to #102

i-make-robots, 2015-05-26:
It seems heirarchy of preferences has changed and it broke way configurations_available is initialized in MachineConfiguration().  Either that or preferences.childrenNames() doesn't behave as it used to.  I had set it up withe [root/machine number/tool name], and preferences.childrenNames() is returning both machine number and all tool name.

i-make-robots, 2015-05-26:
Submitted a fix that filters out all names that don't match the number format.  this is less-than-ideal, as the tool names should not have shown up in this list.  Also on my laptop there are at least 6 machine profiles and I'm only getting one listed, which means the problem could run deeper.  Could other profiles have been wiped somehow?  Are they lost in the twilight zone?

virtuoushub, 2015-05-27:
Not wiped, just residing in a different preference node. I either need to I revert the preference node locations from [com.marginallyclever.makelangelo](https://github.com/MarginallyClever/Makelangelo/blob/dev/java/src/main/java/com/marginallyclever/makelangelo/PreferencesHelper.java#L27) back to DrawBot and the other respective locations or have a tool that maps/moves from the old location to the new.

I have a some tests written for the preferences that recursively prints out the values for a given node. When I get some time I'll open a pull request the aforementioned unit tests. I suspect [Preferences#childrenNames()](https://docs.oracle.com/javase/7/docs/api/java/util/prefs/Preferences.html#childrenNames()) did not change, I probably borked the logic so that [root/machine number/tool name] became [root/tool name]. I am in the process of fixing it so that it was like it was and so that you don't have to filter out names that do not match the number format.

i-make-robots, 2015-05-27:
I know right now when I loads home 551 it only shows me settings for 692. :(

i-make-robots, 2015-05-27:
@virtuoushub, your latest pull seems to fix the issue.  Good job!  I believe the last outstanding issue is to teach me how to properly export so I can update the release package with these fixes.

closed 2015-05-27 by i-make-robots

virtuoushub, 2015-05-30:
@i-make-robots, when I write it the instructions on how to export will be [here](https://github.com/MarginallyClever/Makelangelo/wiki#exporting-a-jar-file).