MarginallyClever/Makelangelo-software #109: Tag v7.1.1 and v7.1.2

virtuoushub, 2015-05-30:
Commit: 6c6e0ade17828995e9e0515d5ad2029253f1e471 should be v7.1.1
Commit: 463c0b9f1b6a16a824fdcaa8708115684e1c6cdf should be v7.1.2

Where a more [formalized process is documented](https://github.com/MarginallyClever/Makelangelo/wiki#exporting-a-jar-file), these types of issues will go away.

i-make-robots, 2015-06-15:
Can i do this from within the github app?

virtuoushub, 2015-06-15:
@i-make-robots, IIRC the GitHub app has a place to launch the terminal, so yes you can run the commands there.

In terms of doing it through the GUI, I use [SourceTree](https://www.sourcetreeapp.com/) and know you can do it through the GUI in that. I have not used the GitHub app in a long time so I can not say for sure. Which version of it are you using(Windows, Mac, ...)?

i-make-robots, 2015-06-29:
Done.

closed 2015-06-30 by virtuoushub