MarginallyClever/Makelangelo-software #110: Recent Line Inaccuracy

chrsmlls333, 2015-06-04:
Noticeable curtaining effect which is new in the past months firmware updates. 
[image]

Has anyone noticed this? The top line is intervals of 10 on the manual controls and the bottom arced line is one movement of 100.

i-make-robots, 2015-06-04:
The firmware splits long lines into short segments to cancel out this
bowing problem.  the MM_PER_SEGMENT value may be too high.  There's also a
chance that drag on the pen tip is causing the line to bend a little bit on
longer movements.

chrsmlls333, 2015-06-04:
Thanks I'll compare the mmPer values of the two firmware versions and get back on if that solves it.

chrsmlls333, 2015-06-21:
Changing MM_PER_SEGMENT from 10 to 3 or 4 fixed this. Easy peasy thanks a lot.

closed 2015-06-21 by chrsmlls333