MarginallyClever/Makelangelo-software #111: Remove Eclipse specific files from project.

virtuoushub, 2015-06-06:
c19ecb52681f1e3b6857627133f1cecdf6cefffe reintroduced the .project file and .settings/ folder

Also, the [classes](https://github.com/MarginallyClever/Makelangelo/tree/dev/java/classes) folder is auto generated and Eclipse specific so it should not really be in the repository either.

These files and folders tend to be specific to one developer's environment.

The Eclipse gitignore template https://github.com/github/gitignore/blob/master/Global/Eclipse.gitignore ignores them and I think it Makelangelo should too.

i-make-robots, 2015-06-15:
ran this on both .project and .settings.  Does that work?

virtuoushub, 2015-06-15:
Yes thanks. 8ad26076cd7fd09f497bbe0046533a2ba06c49a5 did the trick.

commit 8ad26076 (2015-06-15): - ran fix #111
- changed flood fill again for reasons
- improved line weight when zooming
- a bit of house cleaning (unused variables)
PR #114 (2015-06-16): dev v7.1.4-SNAPSHOT
### Changelist:
* Closes #74 
* Minor bug fixes
* Code house keeping

# ~~SIMILAR WARNING AS #108~~

~~To be clear, this PR removes the following files:~~

I left some of the following documentation to capture how to bring back Eclipse specific files.

8ad26076cd7fd09f497bbe0046533a2ba06c49a5 removed the following files:
```
java/.project
java/.settings/*
```

~~I implore you to resolve #111 in one way or the other. My vote is no IDE/developer specific files in the repo.~~

~~If you accept this pull request make sure to have Eclipse closed before you update your Git repo. A [... cut]
closed 2015-06-15 by virtuoushub