MarginallyClever/Makelangelo-software #122: Add missing multilingual support key/values

virtuoushub, 2015-06-30:
I created a [tool to scan for missing translation key/values in the language files](https://github.com/MarginallyClever/Makelangelo/blob/dev/java/src/main/java/com/marginallyclever/util/MarginallyCleverTranslationXmlFileHelper.java). Below are the current results as of 2015-06-30.

I am creating issue to document my findings, however it is possibly dependent on #116 as I am changing the AboutHTMLBeforeVersionNumber, AboutHTMLAfterVersionNumber keys.

## files missing keys:
* [arabic.xml](#arabic)
* [dutch.xml](#dutch)
* [french.xml](#french)
* [german.xml](#german)
* [greek.xml](#greek)
* [italian.xml](#italian)
* [portuguese.xml](#portuguese)
* [russian.xml](#russian)
* [spanish.xml](#spanish)

### <a name="arabic"></a>arabic.xml missing keys:
* [ ] DisengageMotors

### <a name="dutch"></a>dutch.xml missing keys:
* [ ] DisengageMotors

### <a name="french"></a>french.xml missing keys:
* [ ] HilbertCurveName
* [ ] voronoiStipplingDotMax
* [ ] RGBName
* [ ] StatusSoFar
* [ ] voronoiStipplingDotMin
* [ ] SpiralName
* [ ] spraypaintToolDown
* [ ] HilbertCurveOrder
* [ ] CrosshatchName
* [ ] voronoiStipplingName
* [ ] YourMsgHereName
* [ ] DisengageMotors
* [ ] ScanlineName
* [ ] ZigZagName
* [ ] BoxGeneratorName
* [ ] spraypaintToolMaxFeedRate
* [ ] HilbertCurveSize
* [ ] spraypaintToolLiftSpeed
* [ ] StatusRemaining
* [ ] PulseLineName
* [ ] spraypaintToolUp
* [ ] penToolAdjust
* [ ] spraypaintToolDiameter
* [ ] spraypaintToolTest
* [ ] voronoiStipplingCellCount
* [ ] voronoiStipplingGenCount

### <a name="german"></a>german.xml missing keys:
* [ ] LineSegments
* [ ] Diameter
* [ ] PenUp
* [ ] Finished
* [ ] Millimeters
* [ ] Left
* [ ] Bottom
* [ ] ToolType
* [ ] Failed
* [ ] Send
* [ ] PortNotConfigured
* [ ] MachineWidth
* [ ] Right
* [ ] ConnectingTo
* [ ] DisengageMotors
* [ ] Cancelled
* [ ] UpToDate
* [ ] MenuPreferences
* [ ] Centimeters
* [ ] PenDown
* [ ] PortOpened
* [ ] Start
* [ ] Converting
* [ ] Cancel
* [ ] EstimatedTime
* [ ] Unpause
* [ ] MachineHeight
* [ ] Save

### <a name="greek"></a>greek.xml missing keys:
* [ ] HilbertCurveName
* [ ] voronoiStipplingDotMax
* [ ] RGBName
* [ ] voronoiStipplingDotMin
* [ ] SpiralName
* [ ] spraypaintToolDown
* [ ] HilbertCurveOrder
* [ ] CrosshatchName
* [ ] voronoiStipplingName
* [ ] YourMsgHereName
* [ ] DisengageMotors
* [ ] ScanlineName
* [ ] ZigZagName
* [ ] BoxGeneratorName
* [ ] spraypaintToolMaxFeedRate
* [ ] HilbertCurveSize
* [ ] spraypaintToolLiftSpeed
* [ ] PulseLineName
* [ ] spraypaintToolUp
* [ ] penToolAdjust
* [ ] spraypaintToolDiameter
* [ ] spraypaintToolTest
* [ ] voronoiStipplingCellCount
* [ ] voronoiStipplingGenCount

### <a name="italian"></a>italian.xml missing keys:
* [ ] HilbertCurveName
* [ ] penToolLiftSpeed
* [ ] voronoiStipplingDotMax
* [ ] RGBName
* [ ] StatusSoFar
* [ ] voronoiStipplingDotMin
* [ ] SpiralName
* [ ] spraypaintToolDown
* [ ] HilbertCurveOrder
* [ ] CrosshatchName
* [ ] penToolMaxFeedRate
* [ ] voronoiStipplingName
* [ ] YourMsgHereName
* [ ] penToolDiameter
* [ ] DisengageMotors
* [ ] ScanlineName
* [ ] ZigZagName
* [ ] BoxGeneratorName
* [ ] spraypaintToolMaxFeedRate
* [ ] HilbertCurveSize
* [ ] spraypaintToolLiftSpeed
* [ ] penToolDown
* [ ] StatusRemaining
* [ ] PulseLineName
* [ ] spraypaintToolUp
* [ ] penToolAdjust
* [ ] spraypaintToolDiameter
* [ ] spraypaintToolTest
* [ ] penToolTest
* [ ] voronoiStipplingCellCount
* [ ] SharePromo
* [ ] voronoiStipplingGenCount
* [ ] penToolUp

### <a name="portuguese"></a>portuguese.xml missing keys:
* [ ] DisengageMotors

### <a name="russian"></a>russian.xml missing keys:
* [ ] HilbertCurveName
* [ ] voronoiStipplingDotMax
* [ ] RGBName
* [ ] StatusSoFar
* [ ] voronoiStipplingDotMin
* [ ] SpiralName
* [ ] spraypaintToolDown
* [ ] HilbertCurveOrder
* [ ] CrosshatchName
* [ ] AboutHTMLAfterVersionNumber
* [ ] voronoiStipplingName
* [ ] YourMsgHereName
* [ ] DisengageMotors
* [ ] ScanlineName
* [ ] ZigZagName
* [ ] BoxGeneratorName
* [ ] spraypaintToolMaxFeedRate
* [ ] AboutHTMLBeforeVersionNumber
* [ ] HilbertCurveSize
* [ ] spraypaintToolLiftSpeed
* [ ] StatusRemaining
* [ ] PulseLineName
* [ ] spraypaintToolUp
* [ ] penToolAdjust
* [ ] spraypaintToolDiameter
* [ ] spraypaintToolTest
* [ ] voronoiStipplingCellCount
* [ ] voronoiStipplingGenCount

### <a name="spanish"></a>spanish.xml missing keys:
* [ ] HilbertCurveName
* [ ] penToolLiftSpeed
* [ ] voronoiStipplingDotMax
* [ ] RGBName
* [ ] StatusSoFar
* [ ] voronoiStipplingDotMin
* [ ] SpiralName
* [ ] spraypaintToolDown
* [ ] HilbertCurveOrder
* [ ] CrosshatchName
* [ ] penToolMaxFeedRate
* [ ] voronoiStipplingName
* [ ] YourMsgHereName
* [ ] penToolDiameter
* [ ] DisengageMotors
* [ ] ScanlineName
* [ ] ZigZagName
* [ ] BoxGeneratorName
* [ ] spraypaintToolMaxFeedRate
* [ ] HilbertCurveSize
* [ ] spraypaintToolLiftSpeed
* [ ] penToolDown
* [ ] StatusRemaining
* [ ] PulseLineName
* [ ] spraypaintToolUp
* [ ] mmNotice
* [ ] penToolAdjust
* [ ] spraypaintToolDiameter
* [ ] spraypaintToolTest
* [ ] penToolTest
* [ ] voronoiStipplingCellCount
* [ ] SharePromo
* [ ] voronoiStipplingGenCount
* [ ] penToolUp

i-make-robots, 2015-06-30:
>    - arabic.xml <#14e46d7b63dc0215_arabic>
>    - dutch.xml <#14e46d7b63dc0215_dutch>
>    - french.xml <#14e46d7b63dc0215_french>
>    - german.xml <#14e46d7b63dc0215_german>
>    - greek.xml <#14e46d7b63dc0215_greek>
>    - italian.xml <#14e46d7b63dc0215_italian>
>    - portuguese.xml <#14e46d7b63dc0215_portuguese>
>    - russian.xml <#14e46d7b63dc0215_russian>
>    - spanish.xml <#14e46d7b63dc0215_spanish>

Simplest solution?  Add it to english.xml, run the rest through fiverr or
past their original translators.  (good thing I made them put their names &
emails in...)

i-make-robots, 2015-09-23:
This bug will be reopened after this milestone and before the next, as we will surely miss new strings.  Hooray for internationalization!

... wondering if there's a way to curl google translate....

closed 2016-05-18 by i-make-robots