MarginallyClever/Makelangelo-software #124: master branch MachineConfiguration.java:134 exception

i-make-robots, 2015-07-18:
Loading image for machine & paper size panel breaking on some systems.

Show the panel without the picture if it fails?

chrsmlls333, 2015-07-19:
Seems like a good idea for future stability in those dialogs either way.

i-make-robots, 2015-07-19:
The picture doesn't explain itself, doesn't explain the menus that follow, and the whole dialog is something to think about.  Why would you need access to the machine size in order to adjust the paper size?  It doesn't make sense.

chrsmlls333, 2015-07-19:
The idea of starting at the center is also ponderous in larger setups. Having machine size and paper size together is fine I think. Avoids the issue of menu diving. It almost makes me wonder if all the machine settings should be centralized on the new Settings Panel. Reorganization is worth a new discussion topic I think.

virtuoushub, 2015-07-23:
The use of [Class#getResourceAsStream](https://github.com/MarginallyClever/Makelangelo/blob/master/java/src/main/java/com/marginallyclever/makelangelo/MachineConfiguration.java#L133-L134) in MachineConfiguration is not how it is done most other places.  Usually it is done using ClassLoader#getResourceAsStream

I'd be willing to bet this is why it fails sometimes.

i-make-robots, 2015-10-17:
Image removed from 7.3.0 version, not an issue any more.

closed 2015-10-17 by i-make-robots