MarginallyClever/Makelangelo-software #125: Document MarginallyCleverPreferencesHelper

virtuoushub, 2015-07-22:
A while back I wrote a helper class to remove illegal preference nodes(machine names < 0)

@i-make-robots Check out [MarginallyCleverPreferencesHelper](https://github.com/MarginallyClever/Makelangelo/blob/dev/java/src/main/java/com/marginallyclever/util/MarginallyCleverPreferencesHelper.java#L30-L38) if [PreferencesHelperTest]( https://github.com/MarginallyClever/Makelangelo/blob/dev/java/src/test/java/com/marginallyclever/makelangelo/PreferencesHelperTest.java).

The helper class removes all machines whose name is -1 or below.

virtuoushub, 2015-09-09:
This issue and Javadocs are sufficient documentation.

closed 2015-09-09 by virtuoushub

Jip-Hop, 2020-06-11:
I'm trying to get rid of machines with ids < 0, but this solution no longer seems to work. Where is the config stored for each machine? I've been looking everywhere. 😝