MarginallyClever/Makelangelo-software #133: Watermarking

raster, 2015-08-31:
The software now seems to add a watermark. It isn't saved with the G-Code file, but seems hardcoded into the software. Is there a setting somewhere to disable this?

[image]

i-make-robots, 2015-08-31:
It's in the gcode.  There is no disable button atm.

raster, 2015-08-31:
I've exported the G-Code that gets generated and it does not seem to add the watermark (at least from what I can tell running it through CAMotics.) Will there be a preference to turn the watermarking off in the future?

i-make-robots, 2015-09-01:
Gcode is read in verbatim.  Scripts I have built all sign their name except zigzag.

So far you and Joshua are the only people to request it be turned off.  I'm looking into a better customer registration system. Machines we sell could have a disable option.

patlichty, 2015-09-01:
I would second that. A tool should not watermark a work of art; doing so leaves the work invalid. Does my inkjet watermark my prints. Watermarking the output IMO is a curiosity that turns the machine into something other than a tool. I never noticed this because I use zigzag almost exclusively, and have not updated the software since January.

virtuoushub, 2015-09-01:
Artists sign their works. It has not bothered me. I always looked at it as each respective Makelangelo is configured differently and this was a way to show that off.

raster, 2015-09-01:
I checked through things again, and it appears that the watermark is added to the G-Code (my mistake!) which means you can save the G-Code file, edit it to remove the watermark, and then reload it. I suppose that's a workaround for people who don't want a watermark added.

Dan, are you adding these watermarks to each style/script that generates the G-Code? Has anyone else written styles/scripts? Maybe an option would be to ship Makelangelo with a limited number of conversion styles, and then make a set that is available only to your hardware customers, or commercially, or something like that.

I understand that you may want this from a branding standpoint, but as Patrick mentioned, my cameras, and printers, and other tools don't visually watermark the work they output. This would be like MakerBot adding "Made on a MakerBot™" to every print that comes from one of their printers...

i-make-robots, 2015-09-01:
yes, they are added to each style.  no one else has written any styles.
Maybe.

Like I said, maybe a non-branded version for people who bought a
makelangelo.  Register your machine #, get the signature turned off.  Would
mean the machine has to "phone home" more often...  are you ok with that?

raster, 2015-09-01:
I like any option that allows the watermark to be turned off. I really think artists would as well. One of the students I had last semester designed and built his own machine, and was creating beautiful drawings... I'm assuming he wouldn't have wanted watermarked pieces.
I think you've got some good ideas as far as a non-branded version for hardware customers, perhaps with an option to purchase a non-branded version? Or a registration scheme where it phones home. Also, what does the phoning home feature do? Just keep track of how many machines there are? Is other data sent back?

patlichty, 2015-09-01:
> <reply+004b754c07849bf156f2935cf79b2b5e4dc25c68436c67ab92cf0000000111fdb8709
> 2a169ce0634f683@reply.github.com>

Ok, this is something I hadn¹t thought about.  I think that a paid version
of the software, even if modestly priced, is perfectly understandable.
Also, I think for Makelangelo customers proper, I think this should be a
given ­ they¹ve paid for the right to use the tool without watermarking.
There are many free versions of software that are workable with watermark ­
there are many precedents for this.  Good to have clarification.

Patrick Lichty
Assistant Professor of Multimedia
American University of Sharjah
United Arab Emirates

<https://github.com/MarginallyClever/Makelangelo/issues/133#issuecomment-136
829068> .

i-make-robots, 2015-09-01:
All it tracks is the first time a machine turns on.  nothing else, ever
again.
That could be changed to see when a machine last phoned home to give an
idea of how many are regularly active.
Then machines could ask "can I turn off the water mark?"  and my servers
would say yes or no.

patlichty, 2015-09-01:
> <reply+004b754c35bb8d38069255a327ddeeb57437a548557c22ca92cf0000000111fdc4fa9

Think this – This makes a tacit understanding that you will be in business
for the next twenty years.
I’m acting like an engineer working out the worst-case scenario… What
happens if your servers are no longer on?
And yes – I would not be surprised if a nostalgic geek will fire up a
Makelangelo twenty years from now.

Date:  Tuesday, September 1, 2015 at 11:56 PM

<https://github.com/MarginallyClever/Makelangelo/issues/133#issuecomment-1368362
09>

<https://github.com/MarginallyClever/Makelangelo/issues/133#issuecomment-136
841171> .

i-make-robots, 2015-09-01:
if I'm out of business, what do I care?

patlichty, 2015-09-01:
> <reply+004b754c54b6c017dc4b12848e84c032265e4ae6a08ea88192cf0000000111fdc7589

Legacy’s everything; you never know when or if someone might be that geek.
Goodwill’s a good thing, even that far down the line.
Never know when something like that might affect you. I’ve come around the
world and found out that my (probably) ex-girlfriend’s brother-in law is my
HR manager’s best friend in Dubai, and he’s helping me make connections.
Time and space are very small these days, Dan.

Date:  Wednesday, September 2, 2015 at 12:06 AM

https://github.com/MarginallyClever/Makelangelo/issues/133#issuecomment-1368362
<https://github.com/MarginallyClever/Makelangelo/issues/133#issuecomment-1368444
16>

<https://github.com/MarginallyClever/Makelangelo/issues/133#issuecomment-136
845449> .

raster, 2015-09-01:
Will the software still work if it tries to phone home and does not get a response?

As for "why should you care"... Personally, I want my work to outlive me. It's part of what I try to give to the world. I've done this for companies and organizations I've worked for... left them with the tools and information to continue on without me. The programmers who didn't care about 4-digit years caused all the Y2K issues because it was 30 years in the future at the time.

i-make-robots, 2015-09-01:
YES!  It will still work if the database can't be reached.
I understand you want it off.
I'm only pushing back because I'm lazy and I have to figure out a way to
turn it off for paying customers.
When I can see an easy solution it's an automatic yes.
In the meantime I might be able to squeeze out a non-branded copy just for
you.
Super secret, not to be shared, no guarantee of bug-free-ness.

OK?

patlichty, 2015-09-02:
> <reply+004b754c8b01d3c2007e3a74403ae13c8e9ff46a14e69df992cf0000000111fdd4bf9

I totally understand.  Just bouncing around ideas.
That would be great!
Hopefully my #2 will be running in Sharjah in a week or two.
Everyone¹s pretty excited about seeing it work; I hope they see it as having
a process, not just being a curiosity.

Date:  Wednesday, September 2, 2015 at 1:03 AM

<https://github.com/MarginallyClever/Makelangelo/issues/133#issuecomment-1368499
83>

<https://github.com/MarginallyClever/Makelangelo/issues/133#issuecomment-136
860924> .

i-make-robots, 2015-09-02:
Cool cool.  The version I'm sending has all the new bells & whistles.

Are you on mac or windows?

i-make-robots, 2015-10-17:
Adding a "register your machine" option, which will return a key you can use to activate some special features like disabling watermarks.  This way we'll also get a better idea of who are the power users.

raster, 2015-10-17:
Good option!

patlichty, 2015-10-18:
> <reply+004b754cbfbf01cfe3b61e7c1fb1a8feed8ab6e9d8d2605d92cf00000001123a6a589

Great!

Date:  Saturday, October 17, 2015 at 11:58 PM
Subject:  Re: [Makelangelo] Option to disable watermarking (#133)

<https://github.com/MarginallyClever/Makelangelo/issues/133#issuecomment-148
947750> .

i-make-robots, 2015-12-23:
dev version removes all watermarks.

closed 2015-12-23 by i-make-robots

patlichty, 2016-02-05:
> <reply+004b754c36343684c992fe5186a60e6b2e50f8990a7332b292cf000000011292b2389

I¹m going to try to get back to getiing unit 929 working Monday.

Date:  Wednesday, December 23, 2015 at 11:04 PM

Closed #133 <https://github.com/MarginallyClever/Makelangelo/issues/133> .

