MarginallyClever/Makelangelo-software #140: Doxygen?

virtuoushub, 2015-09-29:
Commit 504a5896c8cab3ddcb3f6de0549b685d686e9a5a added the file: `java/Makelangelo.doxyfile`

Is there a reason to use this over Javadoc? ~~(aside from the Javadocs being mostly nonexistent and totally fucked when they are written...)~~ :smirk_cat:

i-make-robots, 2015-09-29:
> being mostly nonexistent and totally fucked when they are written...) [image:

Some people asked for doxygen on another project.  I'll just leave the file
there and let them generate doxygen files when they need to.  generated
files are ignored from the repo.

virtuoushub, 2015-10-08:
Closing per @i-make-robots' [comment](https://github.com/MarginallyClever/Makelangelo/issues/140#issuecomment-144030438).

closed 2015-10-08 by virtuoushub