MarginallyClever/Makelangelo-software #143: Admin privileges prevent Preferences creation; crash on program start

i-make-robots, 2015-10-10:
[image]

virtuoushub, 2015-10-10:
FWIW that warning has nothing to do with the crash before load. That warning is #93. It is just a warning and not a bug. It has only to do with running as a non administrative account on windows. There is no detrimental effect to using the app if you see that warning. See [my previous comment](https://github.com/MarginallyClever/Makelangelo/issues/97#issuecomment-105095394).

Now to the actual cause of the crash. What is the output of:
```cmd
dir
```
?
If `C:\Users\Cataldor2\Desktop\languages` doesn't exist than this is [the reason for the stack trace](https://github.com/MarginallyClever/Makelangelo/blob/dev/java/src/main/java/com/marginallyclever/makelangelo/MultilingualSupport.java#L75-L82).

The multilingual support is trying to load from 
[`System.getProperty("user.dir") + File.separator + "languages"`](https://github.com/MarginallyClever/Makelangelo/blob/dev/java/src/main/java/com/marginallyclever/makelangelo/MultilingualSupport.java#L34) aka `C:\Users\Cataldor2\Desktop\languages`

The app [should be falling back](https://github.com/MakeHartford/Makelangelo/commit/83d549257beed60bacfe7a6c7c35640281135a5f) to use the included [english.xml](https://github.com/MarginallyClever/Makelangelo/blob/dev/java/src/main/resources/languages/english.xml) that is in the jar and not crashing. So I am not sure what that `Makelangelo.jar` looks like but it doesn't seem to be built correctly.

If the language files are included in the jar, and the jar is properly packaged, this issue would never occur [again](https://github.com/MarginallyClever/Makelangelo/pull/126#issuecomment-123957861).

i-make-robots, 2015-10-15:
I probably packaged it wrong.

i-make-robots, 2015-10-17:
This bug is caused by failure to run Maven build script.

closed 2015-10-17 by i-make-robots