MarginallyClever/Makelangelo-software #144: prepare image panel not updating when new machine connects

i-make-robots, 2015-10-15:
I had one machine in my settings; connected a new machine; new machine does everything normally EXCEPT appear in the update panel list of available machines.  this exposes several problems:
1- no update after new machine registers and gets GUID.
2- no update after old machine connects.
3- no way to add a machine manually to edit settings BEFORE connecting

virtuoushub, 2015-10-17:
:+1: 
Machine configuration needs a complete overhaul. When writing the preferences API it was a PITA to work with.

i-make-robots, 2015-10-17:
I'm fine with throwing out prefs and using a cleaner system.  file on disk?  save to cloud?

i-make-robots, 2015-10-17:
What if we ditched prefs completely and serialized machineConfiguration?

virtuoushub, 2015-10-21:
The preferences serialize the machine configuration to disk already, that is not the issue in my mind. My comment was basically a gripe that classes like [MainGUI](https://github.com/MarginallyClever/Makelangelo/blob/master/java/src/main/java/com/marginallyclever/makelangelo/MainGUI.java) and [MachineConfiguration](https://github.com/MakeHartford/Makelangelo/blob/master/java/src/main/java/com/marginallyclever/makelangelo/MachineConfiguration.java) are [God Objects](https://en.wikipedia.org/wiki/God_object).

Having the GUI show that a new machine has been connected is not trivial to add due to the current architecture. I suspect that the issue you were seeing "new machine does everything normally EXCEPT appear in the update panel list of available machines" and you exit the application the new machine will show up in the update panel list of available machines. :+1: :+1:  for it getting fixed.

i-make-robots, 2015-12-14:
Might be fixed by 20247874c69e38fe3bce12298c71561e206c08e4.  Can you confirm?

virtuoushub, 2015-12-29:
@i-make-robots, sorry for the delay. I will check when I get a chance. Most likely, sometime this week.

i-make-robots, 2016-01-16:
Seems fixed here.

closed 2016-01-16 by i-make-robots