MarginallyClever/Makelangelo-software #150: 7.2.2-alpha Release Jar

virtuoushub, 2015-11-01:
I am not sure how [Makelangelo-v7.2.2-alpha-with-dependencies.jar](https://github.com/MarginallyClever/Makelangelo/releases/download/7.2.2-alpha/Makelangelo-v7.2.2-alpha-with-dependencies.jar) was packaged but it does not work for me.

I get the same error as [shadyoaks](https://www.marginallyclever.com/forum/viewtopic.php?f=2&t=5024#p6399). See their screenshot:  [image]

i-make-robots, 2015-11-02:
I used maven build with io7m's Pom tweak.  I tried mavennative first but it wouldn't build.

virtuoushub, 2015-11-02:
If the tag is correct, then you still had my `mavennatives` forked plugin being used ontop of @io7m's fix. Since it [looks like](https://github.com/MarginallyClever/Makelangelo/pull/148#issuecomment-152770607) you are using Java 7, and my fork [only supports building with Java 8+](https://github.com/virtuoushub/mavennatives/issues/5) I am not sure how your build environment behaved when producing the jar. You are going to have to produce another build as the release jar does not seem to work for some users who meet the system requirements.

i-make-robots, 2015-11-02:
I'll have to make sure I'm running 8 and try again.

virtuoushub, 2015-11-09:
To be clear, the dependency plugin with the Java 8 requirement has been removed. Java 8 is not required to package and release the application.

4a5862dda6d3c3193d769297d62e8442b86f4cb3 contains @io7m's fix for native dependencies to address "dependency issues caused by 32/64 bit collision."

A proper packaging of application is.

```bash
./mvnw clean package
```

Closing as the broken release has been documented.

closed 2015-11-09 by virtuoushub

virtuoushub, 2015-12-02:
FWIW I updated `org.colapietro.maven.plugins:nativedependencies-maven-plugin` to work with Java 7. Version 0.2.2+ of that plugin will work with Java 7.

See: https://github.com/virtuoushub/mavennatives/issues/5