MarginallyClever/Makelangelo-software #162: Reduce pen wobble with better pen holder

i-make-robots, 2016-01-16:
Something like http://www.thingiverse.com/thing:798076 but made for timing belt.
Marginally Clever has permission to make a modified version of said thingiverse part and include it in the kits.

devburke, 2016-04-23:
The one thing to be careful about is the long rigid arms. I've been using a Mega Sketch from Brown Dog Gadgets, and it has long acrylic arms that the belts are attached to. Pushed to the extremes of the drawing area, it's been getting distorted. My theory is that the arms through off the trigonometry when the angle they're at gets farther away from the angle between the pen and the steppers. I think it's best to have the belts attach as close to the pen as possible.

i-make-robots, 2016-04-23:
The arms should swivel around the pen.  Got a url for the version you're
using?

devburke, 2016-04-23:
Agreed. I’m thinking of making a replacement for that part.

It’s here: http://www.browndoggadgets.com/collections/robotics/products/mega-sketch

and can be seen up close here https://cdn.shopify.com/s/files/1/0191/2838/products/IMG_5337_1024x1024.jpg?v=1440622913

Someone else built this one and was what introduced me to the idea of these things.

i-make-robots, 2016-04-23:
Wow. That's the same text I used to have on the makelangelo website. Come on, brown dog. Where is the love?

devburke, 2016-04-23:
Ooh, that is rough. These ideas are definitely out there (see http://www.polargraph.co.uk/whats-a-polargraph/), but this ships with Makelangelo firmware, and pulls your text too? I know it’s open source, but some upfront credit would be nice.

devburke, 2016-04-23:
> Up front cash would be even better! :) Tell your friends about my stuff,
> please.

Of course! I hope you don’t mind that I’ve been making my own? I took how open source it all is to mean that you were open to it. It’s both a learning experience for me, and I don’t think I would’ve been in a spot to afford a kit. I have been planning on donating a bit once it’s finished. Hopefully what I lack in funds is somewhat made up for in contribution here :)

i-make-robots, 2016-04-24:
It's open source exactly for those who can't afford it. The next group is people who just want to use a machine and don't want to assemble it. With this robot I make my money on the second group... And sometimes support donations.

i-make-robots, 2016-05-18:
Clear penholder seems much better, I'm quite satisfied.  I consider this issue closed as this pen holder is starting to trickle out to the real world.

closed 2016-05-18 by i-make-robots

Joshua-BrownDogGadgets, 2016-06-10:
Josh from Brown Dog Gadgets.  

Dan, you know full well WHY the text is the same.  Stop playing the "woah is me" card.

We originally started reselling Makelangelo Kits made 100% by you.  Then we started selling hardware SOURCED FROM YOU but with our own acrylic because you refuse to improve your laser cutting or institute the very simple and obvious change of using a CLEAR pen holder.  You know, so people can SEE what they're drawing.  Something we suggested to you two years ago.  

So yes, we are shameless using text that is copy and pasted off your website... because we are reselling kits that we have BOUGHT FROM YOU IN BULK.  

But you're right, we should change it.  The last thing we want to do is hurt your feelings or buy more supplies from.

i-make-robots, 2016-06-10:
@Joshua-BrownDogGadgets are you OK?  I move as fast as I can to act on all the feedback I get and yes, sometimes it can take years.

My issue with identical text is that it hurts your SEO and mine.  Google notices stuff like that.  Thanks for changing it.

Joshua-BrownDogGadgets, 2016-06-10:
Dan,

Your line of comments were insulting.  To anyone viewing them it would
appear as if we were ripping you off, when we have only been using parts
from you to make these machines.  That's what I take issue with. It's a
nice "fuck your" to someone who has spent a decent chunk of money buying
stuff from you.

If it's messing with SEO, just ask us to change it.

I still can't get any machine to work in our shop to work with the new
firmware and software.  Every single one has a motor slipping.  Every time.
Polar graph and core xy.  Looking at the forums it seems I'm not the only
one.  If I can't get it to work in my shop how can I sell it to customers.

I've got a museum in South Carolina who wants 20 machines, but I've put
them on a holding pattern because we can't get thing to work in shop.

Again, all parts are from you.  All software is from you.  I'm growing
tired of spending my evening trying to trouble shoot things that should
work.

Joshua

Joshua-BrownDogGadgets, 2016-06-10:
Whichever motor moves last locks.  The other one is loose.  Happens in both
core xy and polar graph.

Latest version of your software and firmware.

Entirely 100% your hardware.

We had to take our wall mounted unit off the wall as it was the only one
working in the office.  Needed it for a show.

Joshua

i-make-robots, 2016-06-10:
I'm heading into a maker fair this weekend.  I was aware of the issue, and I will continue investigating the cause on Monday.

I have no desire to insult you.  I didn't realize it would come across that way.

Joshua-BrownDogGadgets, 2016-06-10:
Several of us here saw it and it came across that way to everyone.

Let me know how to fix that problem.  I've either got to cancel some
customer orders or figure out a way of fixing things.

i-make-robots, 2016-06-10:
I've been working on it since it was first reported. I'll let you know ASAP.