MarginallyClever/Makelangelo-software #165: Slow drift upward over time?

i-make-robots, 2016-02-17:
Still working on a test case.  Most apparent in DXF files.  Unique feature seems to be lots of long moves with pen up.  Over 10000 commands drift on an A2 drawing will be between 5 and 10 mm, consistent.  Problem is repeatable.  Not related to weight of counterweights, weight of pen, pulley tooth count, or floor() vs lround() in makelangelo-firmware.

Vvasya, 2016-02-23:
I also have this problem.
But I have a slow drift to the right.
10000 commands: 5 - 10 mm

i-make-robots, 2016-02-23:
Please try drawing again with slower acceleration.  What is your test image?

Binky7, 2016-03-25:
I think I am also having this, where the pen looses it's position and the home spot will drift to the right, after a number of commands/lines. This is showing up when I try to overlay multiple drawings on each other, as they don't line up.

I've included a couple of files that highlight the problem, "Free_Dragon_Tatto" draws the outline first, and then fills it in with scales, you can see the scales drifting outside the outline. This takes a while to draw so I've included "12096-NO12HO", which isn't as complicated, if you notice where the wing touches the back, you'll see the error when it is drawn.
[Samples.zip](https://github.com/MarginallyClever/Makelangelo/files/189256/Samples.zip)

The problem is consistent and happens the same way on multiple drawing attempts and for different sizes, so I don't think it is a hardware issue.

The Gcode produced by Makelangelo looks ok in a gcdoe viewer, so I assume the fault is in the firmware.

I did think it was a rounding error, but haven't found one. I redid the maths in the IK() and FK() routines and they seem ok as those motor_line(). I'm now looking to see if I can find anything in the ISRs which is proving to be a bit fiddley, happy for any suggestions if I'm looking in the wrong spot.

BTW. polargraph_line() is passing a z direct to motor_line(), without a cast from float to long.

Also I have noticed that my pen lifter is twitching, so it may be an issue with how long the interrupts are blocked for (but really I'm just guessing).

Binky7, 2016-03-25:
Is timer1 used to PWM the servo, and time the steps for the steppers?

i-make-robots, 2016-03-25:
Timer3 is the ISR, iirc. Pwm and ISR on the same clock? No! Won't work.

i-make-robots, 2016-03-25:
I've wondered for a long time if there's an off-by-one error that's causing drift.  I also notice that pen up seems to increase the drift, but that's not 100% confirmed.  At what speed are you running?

Binky7, 2016-03-26:
I've tried with max feed rate between 3500 and 200, it made no difference, and I've tried dropping the acceleration to 10, and that made no difference.
I've tried setting the number of servos to zero and it made no difference,
I've tried ignoring Z axis changes in the firmware (forcing Z=0), it made no difference,
I've tried ignoring Z axis changes and G04 in the firmware, it made no difference.

I'm not sure it's related to pen up or downs.

And odd thing, I did a drawing using "scanline", it had lots of back and forward and lots of pen up and downing, and yet there was no drift. The problem seems much worse when using dxf files.

i-make-robots, 2016-03-28:
I only see it with DXF files, too.  Perhaps it has something to do with the
way DXF handles curves?  Maybe it has too many places after the decimal?

Binky7, 2016-03-29:
I think it's more to do with the decimal places, here is a gcode file I've been testing with. It will draw multiple triangles on top of each other, it slowly slews off to the left as the home point slides to the left.

It's almost as if the motors get told to take a half step, but decide to take a full step instead.

[Test.txt](https://github.com/MarginallyClever/Makelangelo/files/192656/Test.txt)

i-make-robots, 2016-03-29:
What happens as you reduce the number of decimal places?  Is there a point
where suddenly it's "happy"?

Binky7, 2016-03-29:
No.
I'm now starting to think the problem may be somewhere in the acceleration code.

I removed all decimals, and the drifting problem still happens,
I removed all pen movements from that and the drifting still happens,
I reduce the acceleration to 1, from that and it still happens,
I put acceleration back to 20, but simplified the file so it just continuously drawing a triangle bound by (133,-8),(3,16),(-133,-16). The drift still happens.
I then started continually drawing a straight line from (0,0) to (133,-8) and it's perfect, NO issue, after a few hundred times the lines are still on top of each other. Same is true if I do (3,16) to home or (-133,16) to home.
It is still perfect if I do a straight line back and forth between (133,-8) and (133,-16). Similarly any other line on the triangle is fine, it is just when it draws the whole triangle that the problem starts to show.

i-make-robots, 2016-03-29:
Here's a crazy thought:  count the teeth on your pulley.

We found *one* pulley that said 20 on the side BUT when we counted it there
were *18*.
In every other way totally identical, can't see the diff with the naked eye.

Maybe?

Binky7, 2016-03-30:
Wouldn't any difference in the diameters show up as a curve when trying to draw straight lines? I can draw straight lines without issues.
Also, this problem doesn't seem to be unique to me.

I did check and both pulleys have 20 teeth and the same circumference.

i-make-robots, 2016-03-30:
What happened when you tried reducing the number of decimal places?

This is a really vexxing mystery.

Binky7, 2016-03-31:
As I mentioned, I removed all decimals and the problem persisted.

And yes it is vexing...

i-make-robots, 2016-03-31:
We'll crack it!

What happens if you jog only one motor repeatedly?

Binky7, 2016-03-31:
It jogs back and forth, manages to get back to home every time no matter how much I jog it.

mouseas, 2016-04-09:
I also have this issue, where the pen drifts over time. I have tried it both with Nema-17 steppers, and with 28BYJ-48 geared steppers, and had the same bug.

It seems to happen most often when a line is *almost* perpendicular to one of the strings, so that that motor's steps per second is low but non-zero. Using the Makelangelo Java app to generate g code, I can pretty reliably get it to happen using the "Crosshatch" converter, when it gets to the diagonal lines. "Big Zig Zag" tends to do better, but does sometimes drift.

I've also noticed that, at least with the motor driver board I've been using, that only one motor is held at a time. Anytime one motor steps, the other motor is free to rotate.

My theory so far has been that when the motor tries to step once, and the other motor steps several times, the first motor tends to not turn enough to get to the next step position before the motor is no longer energized. Over time it'll lose steps due to this issue. I've been meaning to switch to using EasyDriver boards, one per motor, so that both motors are controlled independently, so that one motor isn't released when the other has to step. But I haven't gotten around to that project yet, since it will require some significant retooling of the firmware.

i-make-robots, 2016-04-09:
the RUMBA version uses separate drivers for each motor, and are never
disengaged unless a G17 or G18 is sent (only by human demand).

What board are you using?

mouseas, 2016-04-10:
Arduino Uno + a L293D shield (basically a clone of the old Adafruit board). The eBay listing of the one I actually bought: http://www.ebay.com/itm/221798329172?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

i-make-robots, 2016-05-12:
@mouseas your hardware is different and I suspect is a separate issue.  My day is dedicated to finding the cause of this upward drift.

i-make-robots, 2016-05-12:
@Binky7 I ran your test triangle 52 times in a row on A4 paper and I don't get drift.  However I'm still seeing slow upward drift on large DXF files and a small amount on diagonals in crosshatch.  I have no reason to suspect the Java, so i'm looking very hard at the firmware.

i-make-robots, 2016-05-12:
@Binky7 how heavy are your counterweights?

Binky7, 2016-05-14:
So if you can't reproduce it, then it must be in my hardware. I replaced the hardware, steppers, cables, drivers and the problem persists. Also it is repeatable, if I set the same starting point  then I end up with the same errors in the same place every time.

My counterweights are 130 grams

i-make-robots, 2016-05-14:
I'm getting reproducible error, but not the same as you.  My triangle is
fine, but a long DXF is wrong in the same way every time.  My
counterweights are 175g, and I have folded the pulleys like so:

https://www.instagram.com/p/BFRloXKofBi/?taken-by=imakerobots

This way the weights can't hit the floor, but they need to be 2x heavier.
I want to believe this is a software problem.  I don't believe it's on the
Java side.

i-make-robots, 2016-05-16:
This should be a bug in makelangelo-firmware.  I am closing it here and opening it there.

i-make-robots, 2016-05-16:
https://github.com/MarginallyClever/Makelangelo-firmware/issues/5

closed 2016-05-16 by i-make-robots