MarginallyClever/Makelangelo-software #169: Class path contains multiple SLF4J bindings.

i-make-robots, 2016-03-31:
@virtuoushub I tried 

    <exclusions>
      <exclusion> 
        <groupId>org.slf4j</groupId>
        <artifactId>slf4j-log4j12</artifactId>
      </exclusion>
    </exclusions>

in dependencies.  I think Gradle and Maven are fighting each other?

i-make-robots, 2016-04-06:
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/C:/Users/Admin/.gradle/caches/modules-2/files-2.1/ch.qos.logback/logback-classic/1.1.3/d90276fff414f06cb375f2057f6778cd63c6082f/logback-classic-1.1.3.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/C:/Users/Admin/.m2/repository/ch/qos/logback/logback-classic/1.1.3/logback-classic-1.1.3.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.

i-make-robots, 2016-04-06:
Seems to be the multiple bindings to the exact same version.  So what's is the problem?

virtuoushub, 2016-04-22:
Because I ran `gradle init --type=pom` for #167 and the feature is still "incubating" (https://docs.gradle.org/2.12/userguide/build_init_plugin.html#N10D46) it added both [slf4j](https://github.com/MarginallyClever/Makelangelo/blob/master/java/build.gradle#L29) and [logback](https://github.com/MarginallyClever/Makelangelo/blob/master/java/build.gradle#L32).

logback declares slf4j as a dependency (logback is actually a natively implementation of slf4j), so declaring both in gradle seems to be superfluous.

i-make-robots, 2016-05-13:
https://discuss.gradle.org/t/new-to-gradle-seeking-help-please/17591

i-make-robots, 2016-07-19:
No longer happening on my end.

closed 2016-07-19 by i-make-robots