MarginallyClever/Makelangelo-software #174: Start, Start at line #, and Go Home buttons not working in 7.3.4

cruisetheblues, 2016-04-25:
I noticed in the update notes that you changed the program so that the home position must be set before using the "start", "start at line #", or "go home" buttons. However, these buttons do not work even after setting the home position. I could not get the buttons to activate so the software is unusable. 7.3.3 still works fine for me.

i-make-robots, 2016-04-25:
Thank you! Verifying now...

cruisetheblues, 2016-04-25:
Thanks. Let me know if you need any more information. 

Probably not the best place for this but I have a quick question. My X axis is backwards (negative coordinates to the right, positive to the left) but my Y axis is correct. I am currently using the "flip for glass" option to make the print come out correctly but now the picture is mirrored in the GUI. Could a software setting be added which would allow the user to mirror about the x or y axis. I've been looking for a way to correct my problem my modifying the firmware but I haven't been successful.

i-make-robots, 2016-04-25:
Can you build & run the latest developer version?  I'd like to check that it's OK before I make another release.

i-make-robots, 2016-04-25:
The way to fix your mirror problems is to swap the motors.  An alternate solution is to swap the wires at the controller.

cruisetheblues, 2016-04-25:
I've never built a program before but let me see if I can figure it out.

i-make-robots, 2016-04-25:
download makelangelo project.  download Eclipse Java (aka lunar?).  in
eclipse, add maven plugin and then import existing maven project.  finally
run debug.

cruisetheblues, 2016-04-25:
Thanks. I actually just got it running. The buttons work correctly now however when I click start or start at line #, the program continually sends the first command and never moves on. In the monitor window where you see the g code commands it is a continuous stream of the same command. If I click start it sends "N0 D1 L1.8900 R1.8900" forever. If I click start at line # and enter line 19000 It sends "N19000 G00 X-75.xxxx Y123.xxxx" forever. After the first command is sent, the gondola moves to that point correctly and then the computer continues sending the same command. It does not sequence to the next line #.

i-make-robots, 2016-04-25:
Are you running the latest makelangelo-firmware?

cruisetheblues, 2016-04-25:
I believe so but I can certainly flash it right now and see if that fixes it.

i-make-robots, 2016-04-25:
Please do.  I'm running the latest and tested it here with no trouble.

i-make-robots, 2016-04-25:
PS is cruisetheblues your twitter handle, too?

cruisetheblues, 2016-04-25:
No I'm not cruisetheblues on twitter. must be someone else.

i-make-robots, 2016-04-25:
right.  Well.... https://twitter.com/aggrav8d/status/724658066633809920

cruisetheblues, 2016-04-25:
Thanks man. I just appreciate your awesome open source software and you're active support. I'm a mechanical engineer so programming isn't my specialty but I love making robots so it's super helpful to me when people like you allow the use of your software. 

I updated my firmware and my arduino IDE but I'm still getting the same issue. Everything else seems to be working great. I've only been using a single saved file for testing. Let me try something else and see if that makes a difference.

cruisetheblues, 2016-04-25:
Unfortunately the problem still persists. When I use the "start at line #" command This is what is immediately sent.

 M110 N1000 
>  
N1000 G00 X-67.40053 Y-56.827896 
N1000 G00 X-64.75737 Y-56.827896

After which I see

N1000 G00 X-64.75737 Y-56.827896 
>  
N1000 G00 X-64.75737 Y-56.827896 
>  
N1000 G00 X-64.75737 Y-56.827896 
>  
N1000 G00 X-64.75737 Y-56.827896 
>  
N1000 G00 X-64.75737 Y-56.827896 
>  
N1000 G00 X-64.75737 Y-56.827896 
>  
N1000 G00 X-64.75737 Y-56.827896 
>  
N1000 G00 X-64.75737 Y-56.827896 
>  
N1000 G00 X-64.75737 Y-56.827896 
>  
N1000 G00 X-64.75737 Y-56.827896 
>  
N1000 G00 X-64.75737 Y-56.827896 
>  
N1000 G00 X-64.75737 Y-56.827896 
>  
N1000 G00 X-64.75737 Y-56.827896
.... etc

I cant run 7.3.4 because of the issue reported here
When I run 7.3.3 I can only start at the beginning because of Issue 173. Start at line # seems to work great in the latest build. 

7.3.3 works great and I get this

>  
>  
M110 N0 
G00 Z90.0 
>  
>  
N0 M101 T33.0 B-33.0 L-28.2 R28.2 I1 J-1 
N1 D1 L1.8900 R1.8900 
L=33.00,-28.20 
R=33.00,28.20 
Bottom=-33.00  
X0.00 Y0.00 Z90.00 F7500.00  
 
>  
>  
N2 M06 T0 
>  
N3 G00 F3500.0 A20.0 
>  
N4 G00 Z50.0 
>  
N5 G04 P50.0 
>  
N6 G00 X27.753159 Y72.68684 
>  
N7 G00 X25.109999 Y72.68684 
>  
N8 G00 Z90.0 
>  
N9 G04 P50.0 
>  
N10 G00 X25.109999 Y70.043686 
>  
N11 G00 X22.46684 Y70.043686 
>  
N12 G00 X22.46684 Y67.40053 
>  
N13 G00 X25.109999 Y67.40053 
>  
N14 G00 X25.109999 Y64.75737 
>  
N15 G00 X22.46684 Y64.75737 
>  
N16 G00 X22.46684 Y62.114216 
>  
N17 G00 X25.109999 Y62.114216 
>  
N18 G00 X25.109999 Y64.75737 
G00 Z50.0 
.....etc

i-make-robots, 2016-04-25:
So latest build (7.3.5?) is fine, 7.3.4 is broken.  correct?

cruisetheblues, 2016-04-25:
No. The latest build is still hanging for me. It won't advance past the first command in any mode.

i-make-robots, 2016-04-25:
What's your hardware setup?  Also, send an email do dan @ marginallyclever
com if you're interested in designing some hardware.  I have ME questions
that I can't answer on my own.

cruisetheblues, 2016-04-25:
Im running an arduino uno clone with Adafruit Motor shield 2.3. I couldn't get it to connect to the makelangelo software on OS X so I checked the wiki and it seems to be a known issue with clone boards. So instead Im running the software from a Windows 10 VM and it was working great with 7.3.3 except the start at line # wasn't working. 

I have a genuine arduino coming in the mail so I'm hoping that will simplify the system so I can run from OS X. Its possible that my issues are stemming from my complex setup or it could be something I'm doing wrong in building the application.

i-make-robots, 2016-04-25:
Sadly, we can only officially support the kits I make.  I still want to
help you and this might be a problem for our official users, but at a
certain point I have to give up.  I don't even have a 2.3 shield here.

cruisetheblues, 2016-04-25:
I totally understand. I really appreciate your time. If I figure out what's wrong I'll let you know. I'm about to shoot you an email, I'd be happy to help with mechanical design issues.

i-make-robots, 2016-04-25:
I am seeing the same line repeat error here.  Something about the internal
changes I made over the weekend. Working on it....

devburke, 2016-05-05:
Just to chime in, I'm using a clone board and have been able to connect to it from OS X.

i-make-robots, 2016-05-05:
The latest build seems to be OK.

closed 2016-05-12 by i-make-robots