MarginallyClever/Makelangelo-software #176: Page margin setting cropping image

cruisetheblues, 2016-05-01:
When importing an image after using the page margin slider, the image is cropped to fit in the page margin box instead of being resized. This issue was not present in previous releases as far as I'm aware.

i-make-robots, 2016-05-01:
In previous versions the image was sized downwards fit entirely within the
margins.  Now it is sized up to fill the page, tho some cropping may
occur.  In a future version I want to let the user adjust a preview and the
convert the image, so this issue will go away.

i-make-robots, 2017-07-30:
I have internal methods to move and resize an image before it is converted, but I have no UX pipeline to make it happen.  That is to say there is an "open an image" step and a "convert image" step and there should be a "position image" step between the two... but there isn't.  It would require a serious rethink of the system to add such a step.

i-make-robots, 2017-08-17:
Fixed in 4b829857d3eb24dac2be24fa24ca2dbd26ea939c.  I misread this bug.  I thought you wanted to arbitrarily position and scale image before conversion.  It was adding a way to (A) scale the image to fill the paper or (B) fit the image on the paper when I realized what you were talking about.  Wow!  Only 16 months!

closed 2017-08-17 by i-make-robots