MarginallyClever/Makelangelo-software #177: Opens in weird window sizes

devburke, 2016-05-05:
As of 7.4.5, it always opens in a really goofy sized window: http://i.imgur.com/DwJyO44.png
I've run the latest version on Windows at work and I get weirdly sized windows there now too, although not in the same skinny shape. It doesn't seem to remember anything about how it was previously opened, it always opens weird. I guess this should be a separate ticket though.

i-make-robots, 2016-05-12:
Remembering your window size is something new I'm trying in 7.3.  I have improved it in 7.4.5, not sure how well it works now.  Do you have the latest dev branch?

devburke, 2016-05-12:
I don’t currently. I don’t have a Java IDE set up currently, and I just haven’t gotten to doing that, so I’ve just been running off the released jars.

Joshua-BrownDogGadgets, 2016-06-10:
Same thing happens with me.  7.5.

Latest Mac OS X install.

i-make-robots, 2016-06-21:
f90081d15178d984bca04ec90b10024f40755387 may be a fix for this issue.  please let me know how it goes for you.

i-make-robots, 2016-06-23:
ping @Joshua-BrownDogGadgets

i-make-robots, 2016-08-04:
Is this still an issue?  I will reopen if someone confirms.

closed 2016-08-04 by i-make-robots