MarginallyClever/Makelangelo-software #178: Latest versions wasting time on draw

devburke, 2016-05-12:
Previously, the bot would move directly to the next line it wanted to draw. If there was a big region with no lines to draw, it would skip past it and move directly to the beginning of the next line.

On the latest versions, it seems to move back and forth across the entire drawing, whether it's drawing in a particular area or not. This means it takes a lot more time than necessary, but it also means that it risks drying out the markers by having them spend so much time not drawing while they're uncapped.

i-make-robots, 2016-05-12:
DXF files?  It has to do with the order lines are stored in the DXF format.  I just read them in and draw them in that order.  No attempt at re-routing is done.

devburke, 2016-05-12:
Just regular converted jpegs.

i-make-robots, 2016-05-12:
What conversion style?

devburke, 2016-05-12:
Crosshatch. It’s almost exclusively the one I use, cuz it looks great, even if it is slow to draw.

i-make-robots, 2016-05-12:
So it's travelling back and forth over the empty parts of the picture?  I
think I know what that is and I believe it's fixed in dev.  will be in the
7.5.0 release (tomorrow?)

devburke, 2016-05-12:
Yep. It didn’t used to do that, so it must have been a recent change somehow.

devburke, 2016-05-22:
Still having this issue on 7.5.

devburke, 2016-05-22:
Combined with steppers which are moving crazy slow, this makes for some extremely long draw times.

devburke, 2016-05-22:
Strangely, I'm finding that I can't go back to an earlier version either. Even if I flash the Uno, on any version except the latest, the Start button will never activate.

i-make-robots, 2016-05-22:
Must hit home before start will activate.

devburke, 2016-05-22:
Tried that.

i-make-robots, 2016-05-22:
Are you running a stock 3.2 or 2.5.2 kit?

devburke, 2016-05-22:
I'm running the homemade kit, but I believe it's modeled after 2.5.2 (needs to be tethered to a computer).

The latest version works fine, but running an older version, I can't get the start button to activate at all.

i-make-robots, 2016-05-22:
email me your phone number, I'll walk you through how to debug this.

i-make-robots, 2016-06-20:
Is this still happening?  I can't debug a home-made kit as I have no matching hardware.

devburke, 2016-06-21:
Makes sense. Although this one's actually the kit that came from Brown Dog. Either way, I don't have access to it anymore fur a long while. I'm leaving the country for awhile, so at this point it's someone else's worry to get it running right :) I didn't have time to try much. I tried an older firmware, but I spent more of my time figuring out why that other shield wasn't working.

i-make-robots, 2016-06-21:
Brown Dog got their kits from us.  Since then we've started fully
assembling our kits to reduce the number of potential problems.

i-make-robots, 2016-06-21:
@devburke please take a picture of the machine and email it to dan@marginallyclever.com.  I was running tests yesterday on a 2.5.2 and everything seemed fine.  maybe 1/3 as fast as a makelangelo 3.

devburke, 2016-06-22:
Unfortunately, I won't be able to any more. I'm not in the same place any more, and won't be back with the Makelangelo for at least several months.

i-make-robots, 2016-07-01:
Please let me know if it is still wasting time and I will reopen this issue.

closed 2016-07-01 by i-make-robots