MarginallyClever/Makelangelo-software #181: Opening file doens't load it into view

IDmedia, 2016-05-29:
Hi,

Previous version loaded a given image into view/preview by choosing "Open File...", but that doesn't happen anymore. The latest version only gives me a blank screen after loading a file like nothing happened. Is that a known error? Generate art doesn't seem to generate anything either.

i-make-robots, 2016-05-29:
What version are you using, specifically?
Send a URL to the image, and the choice of converter that you are using,
please.

IDmedia, 2016-05-29:
I'm using release 20 from here: https://github.com/MarginallyClever/Makelangelo/releases
Here's the link to the image: http://i.imgur.com/C78pNxk.png

I've tried all converters, but none seem to create a preview. Using Windows 8.1. I'll try to debug this further tomorrow.

IDmedia, 2016-05-30:
Last version that works for me seems to be 7.4.5. However, managed to test the latest version on another PC today and there it works. Is there some kind of cache that may cause this? I've tried to completely reinstall Java without any luck. I'm on a laptop with dual AMD/Intel graphics, but forcing java to run on eiter GPU does't solve it either.

i-make-robots, 2016-05-30:
No cache.  Some preferences that haven't changed since... a long time ago.

i-make-robots, 2016-06-10:
All I can think right now is that the file is failing to open.  Does it silently stop converting before showing the conversion options?

turaluralu, 2016-06-12:
Hi,
I´ve got the same problem too with 7.5 on Mac os X 10.11.5. 
The Computer calculates and Calculates but nothing to see in the preview ....with every Picture.
Cooling van is making noise and a new yellow prompt appears every 3 Seconds....
The same with hilbert curves and all the other art things.

i-make-robots, 2016-06-17:
@turaluralu can you build from source and check if the latest master branch is working better?

turaluralu, 2016-06-18:
Hi Dan,
2 Builds:
the first one was from 1 week ago. Build succeded but the Problem is still there:
build it from Source.  unfortunately the same Problem.
-> opening file.  ->  Done  -> converting......
No preview. The computer is calculating and the cooling van is working hard .... 

Today i took a fresh download from Github and tried to build but as you can see below with no success.
thanks for help!
::
-------------------------------------------------------
 T E S T S
-------------------------------------------------------
Running com.marginallyclever.makelangelo.PreferencesHelperTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 1.724 sec - in com.marginallyclever.makelangelo.PreferencesHelperTest
Running com.marginallyclever.makelangelo.VersionTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.046 sec - in com.marginallyclever.makelangelo.VersionTest
Running com.marginallyclever.makelangeloRobot.MakelangeloRobotTest
Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 0.021 sec <<< FAILURE! - in com.marginallyclever.makelangeloRobot.MakelangeloRobotTest
checkGUID(com.marginallyclever.makelangeloRobot.MakelangeloRobotTest)  Time elapsed: 0.019 sec  <<< ERROR!
java.lang.IllegalStateException: Turn please_get_a_guid to true before release!
	at com.marginallyclever.makelangeloRobot.MakelangeloRobotTest.checkGUID(MakelangeloRobotTest.java:8)

Results :

Tests in error: 
  MakelangeloRobotTest.checkGUID:8 IllegalState Turn please_get_a_guid to true b...

Tests run: 3, Failures: 0, Errors: 1, Skipped: 0

[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 8.966 s
[INFO] Finished at: 2016-06-18T11:57:14+02:00
[INFO] Final Memory: 18M/251M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:2.18.1:test (default-test) on project Makelangelo: There are test failures.
[ERROR] 
[ERROR] Please refer to /Users/turaluralu/Makelangelo/java/target/surefire-reports for the individual test results.
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException

i-make-robots, 2016-06-18:
please_get_a_guid is for internal testing.  I don't want to get a new
machine ID every time I test a device, I want the customer to do it and get
a pin in my global map of machines.  MakelangeloRobot.please_get_a_guid is
a constant that should be false when testing, true when building.

turaluralu, 2016-06-19:
ahh ok, sorry,
build it from masterbranch LATEST but still the same Problem:
Output .:
Opening /Users/turaluralu/Pictures/Rest/REST/1020_0.jpg...
 
Completed
 
Converting /Users/joerg/.m2/repository/com/marginallyclever/Makelangelo/7.5.0/temp.ngc

----------------------------------------------------
and no preview

i-make-robots, 2016-06-19:
What happens when you load temp.ngc into makelangelo (load file)?
Do you see the rest of the robot (pen circle, counterweights, etc)?

i-make-robots, 2016-06-19:
And what is your video hardware?  I'm using (what are now) common opengl
commands.  no wierd shaders etc.

turaluralu, 2016-06-19:
if I try to load the temp.ngc it says: Converting /Users/turaluralu/tmp.ngc    then Opening /Users/turaluralu/temp.ngc and nothing happens further....
I´m using a macbook Pro Notebook with an Intel iris GPU ...very common, I think.
And yes I can see the Rest of the Robot like pen and so on...
It looks quit normal, except the fact that I can´t see a preview....

i-make-robots, 2016-06-20:
Something is failing silently. No warning in eclipse?

turaluralu, 2016-06-20:
ok, installed Eclipse. and load it like described in your documentation.
after  "run as Java Application" following Messgas appeared. I have to say that I am not very familiar with Eclipse....and Java

JOGL> Hello JOAL
Exception in thread "main" java.lang.NoClassDefFoundError: com/jogamp/openal/JoalVersion
	at jogamp.opengl.openal.av.ALDummyUsage.main(ALDummyUsage.java:14)
Caused by: java.lang.ClassNotFoundException: com.jogamp.openal.JoalVersion
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 1 more

i-make-robots, 2016-06-20:
Huh.  I'll have to forward that to the JOGL people.
Thank you!

i-make-robots, 2016-06-20:
Did that happen when you loaded a picture, or when you started the program?

You may need to "Maven > update project" in order to resolve dependencies,
which means you need maven plugin for eclipse installed.

https://github.com/marginallyclever/makelangelo/wiki#ide-specific-steps

turaluralu, 2016-06-20:
wow, that was a fast answer....
It happens when I start the the programm.
I checked if the maven plugin is installed and what should I say ...yes it is.
the maven plugin is called m2me right? It is installed together with the java 8 support for m2me.

turaluralu, 2016-06-20:
I followed the instructions on exactly that Documentation you wrote above "ide specific steps"

i-make-robots, 2016-06-20:
Huh.  So... yeah, back to JOGL people.

i-make-robots, 2016-06-20:
any error specific to when you load an image?  that's the one we're really looking for.

turaluralu, 2016-06-20:
I think there´s a missunderstanding. If I run it from Eclipse (run as Java application) makelangelo won´t start. There is only this message in the console Window.

turaluralu, 2016-06-20:
So, sorry again....I have to select the right "workspace match" I started the wrong one....
now I can start Makelangelo....but when I load an Image following Messages in the console Window:
Could it be that the "," instaed of a "." is the Problem? 

Exception in thread "AWT-EventQueue-0" java.lang.NumberFormatException: For input string: "93,917"
	at sun.misc.FloatingDecimal.readJavaFormatString(FloatingDecimal.java:2043)
	at sun.misc.FloatingDecimal.parseFloat(FloatingDecimal.java:122)
	at java.lang.Float.parseFloat(Float.java:451)
	at java.lang.Float.valueOf(Float.java:416)
	at com.marginallyclever.makelangelo.GCodeFile.estimateDrawTime(GCodeFile.java:136)
	at com.marginallyclever.makelangelo.GCodeFile.loadFromScanner(GCodeFile.java:252)
	at com.marginallyclever.makelangelo.GCodeFile.load(GCodeFile.java:209)
	at com.marginallyclever.makelangelo.GCodeFile.<init>(GCodeFile.java:68)
	at com.marginallyclever.loaders.LoadGCode.load(LoadGCode.java:35)
	at com.marginallyclever.loaders.LoadImage$1.done(LoadImage.java:213)
	at javax.swing.SwingWorker$5.run(SwingWorker.java:737)
	at javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(SwingWorker.java:832)
	at sun.swing.AccumulativeRunnable.run(AccumulativeRunnable.java:112)
	at javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(SwingWorker.java:842)
	at javax.swing.Timer.fireActionPerformed(Timer.java:313)
	at javax.swing.Timer$DoPostEvent.run(Timer.java:245)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

i-make-robots, 2016-06-20:
> at com.marginallyclever.makelangelo.GCodeFile.(GCodeFile.java:68)

absolutely.  There should never be a comma in gcode AFAIK.
Please send me a URL or dropbox link to the temp.ngc file and I can look at
it myself.

turaluralu, 2016-06-20:
is it possible to send you a mail?

i-make-robots, 2016-06-20:
Absolutely.  dan@marginallyclever.com

turaluralu, 2016-06-20:
done :-)

turaluralu, 2016-06-20:
I changed every "," with a "." in that temp.ngc file and it works.....  Hurray....

i-make-robots, 2016-06-20:
Thank you for your help.  I think I found it. 
Your computer thinks that in your area decimal numbers should be written with a comma.  This breaks the Makelangelo import.  I have just pushed an update to the master branch that tells Makelangelo to cut that out.  Please get the latest from github, run the app, convert your picture again, and let me know the results.

turaluralu, 2016-06-20:
Ok I´ll trie (tomorrow) it´s a little bit late now because I am in Germany.
not you have to thank ….I have to thank you….
I will give a feedback tomorrow as soon as possible.
many many thanks for your help, Joerg

turaluralu, 2016-06-20:
Hi Dan,
I could´t wait so I tried. 
Unfortunately it did´t work.
ngc with commas again but the message in 
the makeangelo „console" is:

       Opening /Users/joerg/Makelangeloo/java/temp.ngc...

10350 line segments.
89.49832cm
Estimated draw time: 58m8s .

Completed     

A bit more than before. But no preview.

the Eclipse Output:

for today it´s enough I think. 
would be nice to hear from you next time. 

Tired, Joerg

i-make-robots, 2016-06-20:
I missed a word in 951bb6a1627c9580046462b088ab59433e4eb968.  One more update from github should do it (I hope).

turaluralu, 2016-06-21:
It´s done!!! It works...what should I say? Thank´s for your help.

i-make-robots, 2016-06-21:
Tell your friends?  Let Brown Dog know?

i-make-robots, 2016-06-21:
@IDmedia does this patch fix your problem with previews?

i-make-robots, 2016-06-21:
Never mind!  Re-read the conversation.  Glad everyone is happy.

closed 2016-06-21 by i-make-robots

turaluralu, 2016-06-21:
yes, it does....I told you above ...It works!!!! Thank you very much....