MarginallyClever/Makelangelo-software #182: Steppers Don't Lock

Joshua-BrownDogGadgets, 2016-06-10:
Using latest firmware and software.

Only one Stepper ever "locks" in place, whereas the other spins freely.  

Drawing does not work.  Even standing still doesn't work as whichever side is "spinning freely" will be dragged up by the counter weight.

The side that stays locked is always the last side which moved.  Meaning if the left stepper moved last, it will stay locked.  If the right stepper moved last, its locked.

This has happened with four different sets of hardware entirely sourced from Marginally Clever.

i-make-robots, 2016-06-10:
I've just built from scratch and retested three makelangelo 2.5.2 running Makelangelo software (release 20) and firmware (latest master branch).  I cannot reproduce the problem at the moment.

Please provide a more specific use case.  Maybe a picture of your hardware setup?

i-make-robots, 2016-06-17:
https://github.com/MarginallyClever/Makelangelo-firmware/issues/8  If problem persists, please post in this other thread.

i-make-robots, 2016-06-17:
For future readers: this is a firmware problem, not a software problem.  See link above.

closed 2016-06-17 by i-make-robots