MarginallyClever/Makelangelo-software #183: Core XY Messed up in 7.5

Joshua-BrownDogGadgets, 2016-06-10:
Core XY gets super confused in 7.5.  The software doesn't know where it "is" on the paper.  

Hitting both "bottom" and "top" take you in the same direction, as the software things the pen is "below" the paper, not above the paper.

Resigning home doesn't actually work.

This could be an issue with things in Polargraph mode as well, since I read somewhere that "home" has been moved to the Top of the page, as opposed to the Center of the page.  (Which is kind of silly if you're using different sized paper in different draws.  Center makes sense.)

i-make-robots, 2016-06-10:
The home position is now fixed to one edge.  It's numerically the same thing to the machine, but now you have a better chance of restarting a drawing because the home is not hidden under the paper.  It also means you can use the template included in \calibration pattern\ to more easily find home, instead of measuring everything with a ruler and making messy marks on your board.  (especially useful when drawing on glass.)

The only difference between the two (polargraph and corexy) is the IK() method, which has not changed.  I added a testKinematics() in the setup() for the latest version, which would spam the serial with error messages if FK(IK(a))!=a.  https://www.instagram.com/p/BGdh0MUA5SP/?taken-by=funvill is currently using the CoreXY code and reporting success.

i-make-robots, 2016-06-13:
I have pinged funvill to see if he's made any changes to the corexy firmware, and if he has, to send a pull request with his changes.

i-make-robots, 2016-06-14:
Does this help? https://github.com/funvill/Makelangelo-firmware/commit/7fad5f2f00d4b74c649420769940e51b874eb79e

i-make-robots, 2016-06-21:
Ping @Joshua-BrownDogGadgets

i-make-robots, 2016-07-19:
I'm closing this issue.  Please let me know if the problem persists and you want I should reopen it.

closed 2016-07-19 by i-make-robots