MarginallyClever/Makelangelo-software #184: UI CHANGE NEEDED: Add a "Reload last Image" button

Joshua-BrownDogGadgets, 2016-06-10:
The UI needs an "Reload last image" button.  

During normal use people frequently wish to change a setting or style if it didn't meet their needs the first time.

It's needlessly tedious to have to go back and find the file again. This is especially handy if you have a folder with a very large number of images in it.  Having a "Last Image" button would make things much easier.  

This was a Menu option in older versions of the software, and it was quite handy.  We're all sad to see it go in newer versions.

i-make-robots, 2016-06-10:
I had no evidence that it was a desired feature.  That's good to know.  I'll see about putting it back in.

i-make-robots, 2016-06-13:
This morning I looked into how to do this.  I'm disappointed it will mean an extra layer of popups - one to say "here is the last file you opened [pick a new one] [next step]"

Joshua-BrownDogGadgets, 2016-06-13:
Or you could just add a button.  Something that says "Reload Last Image".

No need for pop ups.  It would make choosing a design or tweaking designs a
whole lot easier.

From our normal use of the software at shows, exhibits, and in our offices
this button would be frequently used.  Far far more than the "Generate Art"
button, for instance.  (We rare use that except for tests.)

i-make-robots, 2016-06-13:
Please grab the latest master branch, test the change, and let me know if it works for you.

closed 2016-06-14 by i-make-robots