MarginallyClever/Makelangelo-software #198: Engage/Disengage (again)

tjhawkey, 2016-08-27:
Hi Dan, seems to me like it makes the most sense to have the motors disengaged when things start up, engaging when you set home, and then disengaging when a print is complete.
That way, it's easy to move the pen to the home position and saves the motors (and my ears) when a print is complete.

i-make-robots, 2016-08-28:
I hear you, @tjhawkey.  Can you submit a patch that correctly enables
motors at all appropriate times?  I switched back because it was such a
nuisance.  I like using the manual controls to drive to home.

tjhawkey, 2016-08-29:
Hi Dan, I'm not much of a programmer, but I'll take a look.The latest AMS firmware works perfect for at the end of a print--it holds briefly then releases and the motors shut down.I hadn't considered using the manual controls to go to the home position. I'll try that first.Thanks!Tim

i-make-robots, 2016-09-05:
@tjhawkey do you still consider this an issue?  I'd like to close this if you're happy.

tjhawkey, 2016-09-06:
Sure. It works how you intend it.Thanks! It's super fun.

closed 2016-09-07 by i-make-robots