MarginallyClever/Makelangelo-software #207: DXF output is broken or non-standard

ShacharWeis, 2016-09-10:
The DXF file that is exported from the latest dev fork either broken or non-standard. 
These applications wont open it:

Rhino3D
InkScape
CamBam

The only thing that did manage to open it was this online tool:
dxfconverter.org

i-make-robots, 2016-09-10:
Any idea why it's broken?  I can open it in Inkscape.  Maybe that app is very forgiving.

ShacharWeis, 2016-09-10:
My bad, InkScape does open it, just takes forever doing so.

i-make-robots, 2016-09-10:
Please grab the latest again.  I found a bug in FloatField that was rounding down.  If you pen tip size was <1.0 it would cause an infinite loop and nothing would save.

i-make-robots, 2016-09-10:
Too much data :)

ShacharWeis, 2016-09-10:
If I open it in InkScape and export it as DXF than Rhino and CamBam will open it. Maybe you can use that to compare the differences in the files.

i-make-robots, 2016-09-10:
I don't have rhino or cambam. Can you do it and send me the files?

i-make-robots, 2016-09-12:
I'm leaving this as won't fix until I get files I can compare.  Feel free to reopen the ticket when that's available.

closed 2016-09-12 by i-make-robots