MarginallyClever/Makelangelo-software #212: 7.8.3 doesn't save width settings for custom machine

tjhawkey, 2016-11-05:
This did work in 7.8.1. About to test 7.9.0

i-make-robots, 2016-11-05:
Can you write a java unit test that would confirm this always works?

tjhawkey, 2016-11-05:
Doesn't save settings in 7.9.0 for custom machine, 2+ or 3+. Always wants to be 914.4 mm wide and 1219 tall. Makelangelo 5 is un-adjustable 650 x 1000.

Sorry, I'm not clear on what kind of java test you need. My wife is the programmer. :)

i-make-robots, 2016-11-05:
M5 should be. 2 and 3 should be adjustable. There should be a unit test to keep this from breaking.

i-make-robots, 2016-11-05:
There is a patch in the master branch.  Can you test it and see if that helps?

tjhawkey, 2016-11-05:
Yes.
Thanks!

⁣From my phone

Sent from BlueMail

​

tjhawkey, 2016-11-10:
Hi Dan, I am still working on learning how to make a jar for myself.If you send me one, I can do the test right away. The new features look great and I'm excited to try them once I can put the size in.Thanks,Tim

i-make-robots, 2016-11-10:
Can you follow the github wiki steps? Once you have eclipse set up click the green big to run.

i-make-robots, 2016-11-11:
@tjhawkey any news?

tjhawkey, 2016-11-11:
Hi Dan, I plan to test this weekend.
Thanks!

⁣From my phone

Sent from BlueMail

​

turaluralu, 2016-11-22:
Hi Dan,
I have had the same Problem saving the settings. Build it from scratch (masterbranch) .
Now it works.
Thanks a lot, J.

turaluralu, 2016-11-22:
Hi again,
aarrrghh now I can´t save custom paper settings.
Another thing is that all settings can only be saved when the last change  is the Paper setting to anything (exept custom) 
j.

i-make-robots, 2016-11-22:
please list the steps you're taking so i can repeat them and see the effect.

tjhawkey, 2016-11-23:
Hi Dan, I had previously set the machine size in 7.8.3 for custom machine. No problem.7.9.0 has (as you know) options for 2+, 3+, 5+, and custom.It remembers paper color, and paper size (and I think everything else on those two tabs), but won't retain machine size.I can set it and save it, but when I go back, they are still the default values, 914.4 and 1219.2Pulley size and direction are retained properly.Thanks, I'm having great fun drawing pictures of my family.Tim

tjhawkey, 2016-11-23:
Also, I haven't sees any evidence of the other behavior relating to the order of changes. Tim

i-make-robots, 2016-11-23:
> auth/ABZYhnTL-5BaPNzzT2J2za0Ad-BBRA_rks5rA2dPgaJpZM4KqHIJ>> .>-- Dan

and you're building from the dev branch?  I thought I fixed this bug two
weeks ago.

tjhawkey, 2016-11-23:
You are right (I hope).  I haven't had time to build yet (sorry).  I coach a FIRST Lego League team and we're getting ready for our tournament.Tim

i-make-robots, 2016-11-23:
I make robots and I'm getting ready for Black Friday while filling back
orders.  The gift of success!

turaluralu, 2016-11-23:
Hi Dan,
you : please list the steps you're taking so i can repeat them and see the effect. 
settings, changes Paper Size to custom 420 x 560, save . If i klick settings again the Paper Size has not changed.
Same thing at any other settings. But here I can work around if the last step is changing Paper Size to what ever (except custom)  than all the other Settings are saved too. 

You: and you're building from the dev branch? I thought I fixed this bug two weeks ago. 

Builded main Branch. not dev branch. Maby that´s the Problem. I´ll try.....

turaluralu, 2016-11-23:
Hi,
build from dev branch is not good for me. No settings can be saved  :-(
So I´ll build it from master branch again and try to put my DIN A2 Paper (420x560) in PanelAdjustPaper.java
 and compile again. It´s a workaround I kow but maby it works for me.

i-make-robots, 2016-11-23:
I tried your steps, paper size settings are saving for me.

turaluralu, 2016-11-23:
hmm, maybe it's a problem regarding my german keyboardlayout.  
How ever. The workaround, modifying the PanelAdjustPaper.java works fine and I´m happy.
Thanks for your help, J.

i-make-robots, 2016-11-23:
german keyboard shouldn't make a difference.  are you typing comma ","
instead of period "." for decimals?  I thought I took care of that....

turaluralu, 2016-11-23:
tried both and without decimal separator. But saving isn´t possible.

i-make-robots, 2017-07-30:
Is this issue still ongoing?

tjhawkey, 2017-07-30:
Hi Dan, I think the non-saving is resolved. At least the software on the
machine saves it, though I haven't tested it with my custom makelangelo yet.
I will try with the hardware later this week.
Thanks

closed 2017-09-29 by i-make-robots

Ianmcmill, 2019-04-25:
On 7.20.0 upon disconnecting the custom settings will get discarded and custom settings revert back to 650 width and 1000 height.

i-make-robots, 2019-04-25:
Is this makelangelo #0?  If it has a custom of number then it will remember.

i-make-robots, 2019-04-25:
Edit: SHOULD remember.

Ianmcmill, 2019-04-25:
Yeah it does now somehow. Don't know why I didn't earlier. Maybe a java thing or my fault.