MarginallyClever/Makelangelo-software #216: Compiling firmware for Rumba warning

cojarbi, 2017-03-29:
Downloaded the firmware, loaded it on IDE 1.8.2 OSX. With no changes made to the code i run the compile but get the following warning. It does compile and upload

/Users/ayahni/Downloads/MarginallyClever-Makelangelo-firmware-c86acee 2/firmware_rumba/firmware_rumba.ino: In function 'float parseNumber(char, float)':
/Users/ayahni/Downloads/MarginallyClever-Makelangelo-firmware-c86acee 2/firmware_rumba/firmware_rumba.ino:583:13: warning: ISO C++ forbids comparison between pointer and integer [-fpermissive]
   while(ptr>1 && *ptr && ptr<serialBuffer+sofar) {  // walk to the end

i-make-robots, 2017-03-29:
while(ptr>(char*)1 && (*ptr)>0 && ptr<serialBuffer+sofar) ?

cojarbi, 2017-03-29:
Yep that fixes the isuue

i-make-robots, 2017-03-29:
please try the dev branch and let me know if that works for you, the change was already submitted on that branch.  if you confirm it's ok, I will make it the master branch for everyone.

i-make-robots, 2017-03-29:
nm, committed.

closed 2017-03-29 by i-make-robots