MarginallyClever/Makelangelo-software #217: Compiling firmware for ramps warning

cojarbi, 2017-03-29:
Exactly the same procedure done in #216 but uncommented 
// Your choice of board
//#define MOTHERBOARD BOARD_RUMBA
#define MOTHERBOARD BOARD_RAMPS
//#define MOTHERBOARD BOARD_SANGUINOLULU

I get the following warning but it does compile

In file included from /Users/ayahni/Downloads/MarginallyClever-Makelangelo-firmware-c86acee 2/firmware_rumba/firmware_rumba.ino:13:0:
sketch/configure.h:106:0: warning: "MOTHERBOARD" redefined
 #define MOTHERBOARD BOARD_RAMPS
 ^
sketch/configure.h:100:0: note: this is the location of the previous definition
 #define MOTHERBOARD BOARD_RUMBA
 ^
/Users/ayahni/Downloads/MarginallyClever-Makelangelo-firmware-c86acee 2/firmware_rumba/firmware_rumba.ino: In function 'float parseNumber(char, float)':
/Users/ayahni/Downloads/MarginallyClever-Makelangelo-firmware-c86acee 2/firmware_rumba/firmware_rumba.ino:583:13: warning: ISO C++ forbids comparison between pointer and integer [-fpermissive]
   while(ptr>1 && *ptr && ptr<serialBuffer+sofar) {  // walk to the end

i-make-robots, 2017-03-29:
> sketch/configure.h💯0: note: this is the location of the previous

only one MOTHERBOARD line should be uncommented.

cojarbi, 2017-03-29:
Only one was uncommented as displayed above. Could there be a redundancy for rumba somewhere else?

i-make-robots, 2017-03-29:
it's telling you in the error message where the two lines are.  "this is the location of the previous definition".

cojarbi, 2017-03-30:
Making the following changes makes it compile without without issues but not sure if its ok. 
Comment line 100
Uncomment line 106

#if MAKELANGELO_HARDWARE_VERSION == 5
#define USE_LIMIT_SWITCH    (1)  // Comment out this line to disable findHome and limit switches
#define MOTHERBOARD BOARD_RUMBA
#endif
#if MAKELANGELO_HARDWARE_VERSION == 3
#undef USE_LIMIT_SWITCH
//#define MOTHERBOARD BOARD_RUMBA
#endif

// Your choice of board
//#define MOTHERBOARD BOARD_RUMBA
#define MOTHERBOARD BOARD_RAMPS
//#define MOTHERBOARD BOARD_SANGUINOLULU

Furthermore in the Desktop app no port will be shown but the OS does show it and repetier host for example connects to it. 
All this was tested making sure the connection was not open by another app

[image]
[image]
[image]

i-make-robots, 2017-07-11:
Hrm.  Most of these apps (like repetier) use the same libraries that I do to connect over serial.  I don't know why your USB list would be empty.  Try running as administrator.  It might be that Java security doesn't let you access serial ports - sliiiiiightly overprotective.

i-make-robots, 2017-07-30:
Did you solve this issue?

closed 2017-08-01 by i-make-robots