MarginallyClever/Makelangelo-software #219: start windows.bat fails, seemlingly due to unescaped closing paren

rparrett, 2017-07-10:
Downloaded latest release bundle from
https://github.com/MarginallyClever/Makelangelo-software/releases/download/7.11.0/Makelangelo.zip

Windows 7 Pro 64-bit
Java 1.8.0_131 x86_64

Outputs

```
64 bit Windows
ERROR: The system was unable to find the specified registry key or value.
64 bit Java installed.
Java was unexpected at this time.

C:\Users\rparrett\Downloads\Makelangelo>
```

"Java was unexpected at this time." seems to be a batch file syntax error.

Changing

`  @ECHO Java not found.  Please recheck (or reinstall) Java and try again.`

To

`  @ECHO Java not found.  Please recheck (or reinstall^^^) Java and try again.`

Fixed things for me:

```
64 bit Windows
ERROR: The system was unable to find the specified registry key or value.
64 bit Java installed.
The current Java runtime is 1.8
finding the latest Makelangelo JAR...
Found Makelangelo-7.11.0-with-dependencies.jar
```

I'd submit a PR, but I don't think these batch files are tracked in the repository.

i-make-robots, 2017-07-11:
That's odd.  Why would ^^^ fix things?

i-make-robots, 2017-07-11:
Oooooh, the () inside the other ().  Doh!  Removed interior () and updated.  Thank you!  These files used to be tracked and it confused folk.  Thank you for your help!

closed 2017-07-11 by i-make-robots