MarginallyClever/Makelangelo-software #221: Problem displaying CMYK conversion options screen

dentedachiii, 2017-08-08:
As briefly discussed on instagram, this was my first time messing with compiling the dev branch. I got it to compile and run, and can generate CMYK. So much faster than manually editing individual C, M, Y,  and K files in Photoshop. Cant test it yet, as my machine is not set up. 

The problem I ran into was the formatting on the CMYK options screen. Cant really see the text box and description for editing the number of dithering passes. Not sure if Im doing something wrong, though. See attached (literal) screen shot. 
[image]

dentedachiii, 2017-08-08:
not sure why that image appears upside down here...

i-make-robots, 2017-08-08:
Attempted fix in 8f4e4254283a95c8d45035f7150950a026345a3e

commit 8f4e4254 (2017-08-08): Attempt to fix #221

added return command to split message into mutliple lines
closed 2017-08-17 by i-make-robots