MarginallyClever/Makelangelo-software #222: tell me how to make the pen up and down using z axis

Davidkillua, 2017-08-26:


i-make-robots, 2017-08-26:
G0 Znnn change the pen angle.  Settings > pen > up/down test will also work.

closed 2017-08-26 by i-make-robots

Davidkillua, 2017-08-26:
i'm using my 3d printer to do the pen up and down
even though explain more b'cus i'm new to this

Davidkillua, 2017-08-26:
i changed the g code from inkscape but my z axis don't respond 
i want to make pen up down by moving z axis not by moving the pen

i-make-robots, 2017-08-26:
Please send a picture of your machine with the pen holder.  I want to write
a post that includes all the steps, your machine will be the main picture.

i-make-robots, 2017-08-26:
gcode from makelangelo uses Z for z axis, same as a regular cnc machine.

i-make-robots, 2017-08-26:
> Ph: +1 (604) 259-9564 <(604)%20259-9564>
> Mo: +1 (604) 916-2281 <(604)%20916-2281>

https://www.marginallyclever.com/forums/topic/makelangelo-software-drawings-on-a-3-axis-gantry-cnc-robot/
post a picture of your machine there, and we'll continue the conversation
where more people are likely to see it and offer help.

Davidkillua, 2017-08-26:
what should i do
my main thing is to make that inti a cnc router
with the drill tool

Davidkillua, 2017-10-07:
hi
dan
 i need your help
i got
two stepper motor
gondola for vertical plotting
servo motor for pen up and down
adurino uno and cnc shield as controller
so guide me on this to work
i wanna make vertical plotter

Davidkillua, 2017-10-07:
i  need connections guide also

i-make-robots, 2017-10-07:
1. This is the github repo for software support. You are asking for hardware support.

2. I charge $200/h for consulting. If you want free help, try the project forums at marginallyclever.com.

Davidkillua, 2017-10-07:
Oh then can you tell me where to connect this

i-make-robots, 2017-10-07:
https://www.marginallyclever.com/forums/forum/support/