MarginallyClever/Makelangelo-software #225: Settings change, Number format exception

Nakio195, 2017-12-20:
When changing the settings of paper tab or pen tab the application throw an NumberFormatException and deny changes on Windows 7, Makelangelo 7.15.0
Java version 1.8

Here is the log : 

```
Exception in thread "AWT-EventQueue-0" java.lang.NumberFormatException: For input string: "210,0"
        at sun.misc.FloatingDecimal.readJavaFormatString(Unknown Source)
        at sun.misc.FloatingDecimal.parseDouble(Unknown Source)
        at java.lang.Double.parseDouble(Unknown Source)
        at java.lang.Double.valueOf(Unknown Source)
        at com.marginallyclever.makelangeloRobot.settings.PanelAdjustPaper.save(PanelAdjustPaper.java:291)
        at com.marginallyclever.makelangeloRobot.settings.MakelangeloSettingsDialog.run(MakelangeloSettingsDialog.java:97)
        at com.marginallyclever.makelangeloRobot.MakelangeloRobotPanel.actionPerformed(MakelangeloRobotPanel.java:449)
        at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
        at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
        at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
        at java.awt.Component.processMouseEvent(Unknown Source)
        at javax.swing.JComponent.processMouseEvent(Unknown Source)
        at java.awt.Component.processEvent(Unknown Source)
        at java.awt.Container.processEvent(Unknown Source)
        at java.awt.Component.dispatchEventImpl(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Window.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)
```

The 210.0 number is the field for paper width (A4).

Nakio195, 2017-12-20:
It seems that this issue has been covered by drewler in it's commit https://github.com/MarginallyClever/Makelangelo-software/pull/224

i-make-robots, 2017-12-20:
Thank you for this report.  Huh.  I thought I took care of all EU number formatting issues a while ago.  And how did I never see this pull request before? :(  Thank you, @drewler & @Nakio195

i-make-robots, 2017-12-21:
@Nakio195 Are you able to test the latest dev branch?

Nakio195, 2017-12-23:
Tested and the issue is corrected ! Thank you both !

closed 2017-12-23 by Nakio195