MarginallyClever/Makelangelo-software #230: Conversion options never come up

drum303, 2018-02-14:
I am working with version 7.15.3.  It had worked previously.  It one point during a drawing I clicked the halt button, then after that loading an image never brings up the conversion options window.  Command window sits at java.awt.EventDispatchThread.run(Unknown Source) and just sits there.  Version 7.15.0 works as expected.  The same problem exists on 3 different PCs.  I tried removing the machine ID from the registry and wiping out the data in the temp folder in case there was corruption there but no luck.  Again version 7.15.0 works just fine on all these machines.

drum303, 2018-02-20:
It seems to have something to do with the boxy filter.  I can recreate the issue by opening an image, selecting the "boxxy" filter and it never finishes rendering.  Killing the process and restarting it keeps the conversion box from coming up, possibly because it attempts to complete the boxxy filter upon restarting.  This happens in 7.15.4 as well, but 7.15.0 is fine.

i-make-robots, 2018-02-28:
Fixed in release 34, v7.15.5.  Also added a unit test to make sure it can never happen again.

closed 2018-02-28 by i-make-robots