MarginallyClever/Makelangelo-software #233: SVG Generation

jsecondo, 2018-05-30:
I'm having trouble while save the work.

DXF format seems to work fine (size of the file grows with complexity of the designs, but I can't confirm since I don't use the format).

Gcode is well generated.

When I try SVG 1.1 it always generates a text file of 244 bytes of length. 

<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg xmlns="http://www.w3.org/2000/svg" version="1.1" viewBox="0 0 1485.0 2100.0">
</svg>

And that's all folks. 

Tried version 7.15.6, ...15, ....14 all did the same.

jsecondo, 2018-05-30:
Here goes the contect.

> <?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg xmlns="http://www.w3.org/2000/svg" version="1.1" viewBox="0 0 1485.0 2100.0">
</svg>

jsecondo, 2018-05-30:
`<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg xmlns="http://www.w3.org/2000/svg" version="1.1" viewBox="0 0 1485.0 2100.0">
</svg>`

jsecondo, 2018-05-30:
"<?xml version="1.0" encoding="UTF-8" standalone="no" ?>"
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg xmlns="http://www.w3.org/2000/svg" version="1.1" viewBox="0 0 1485.0 2100.0">
</svg>"
[paleta1.svg.txt](https://github.com/MarginallyClever/Makelangelo-software/files/2052253/paleta1.svg.txt)

jsecondo, 2018-05-30:
[image]

i-make-robots, 2018-05-30:
Have you tried the development version?  I believe it's fixed in there and
I'm overdue to publish an official update.

jsecondo, 2018-05-30:
Will try that as soon as I get home.

Thanks

i-make-robots, 2018-06-05:
Bug found and fixed.  Release v7.16.1 resolves the issue.  Thank you, @jsecondo !

closed 2018-06-05 by i-make-robots

jsecondo, 2018-06-06:
You are very welcome. KEEP UP THE EXCELLENT WORK!!!

jsecondo, 2018-06-08:
Tried version 15.6.1. Now the SVG generates code but when I try to open the SVG file, I can see HALF the design which is located all the way to the right.
Tried with this image. 3 SVG files in the zip file.

[image]
[IronMan.svg.zip](https://github.com/MarginallyClever/Makelangelo-software/files/2082529/IronMan.svg.zip)

i-make-robots, 2018-06-08:
I’m not home atm. Have you tried in inkscape? I loaded my test images in internet explorer and they were fine.

jsecondo, 2018-06-08:
[image]

Same effect. Is this normal? Don't know about SVG settings or configuration. Can it be parametrized using the width and height of the machine?

i-make-robots, 2018-06-09:
separate issue #236 created.