MarginallyClever/Makelangelo-software #236: SVG export not centered to paper, paper size wrong

i-make-robots, 2018-06-09:


i-make-robots, 2018-06-09:
@jsecondo if you can try the developer version and confirm?  Also try with landscape paper rotation.  paper in inkscape should make machine size in makelangelo, drawing should be centered.

jsecondo, 2018-06-09:
Changing to landscape works quite alright. Did recentering on the page by hand.

The SVG I was trying was a CMYK. The SVG contains all the lines but doesn't discriminate on the color, right? So to have a SVG of each color component I have to do the separation and then treat each component separately. Do I have a way to replicate each color drawing from the CMYK style? Guess that in GCODE I'll find the way.

jsecondo, 2018-06-09:
Did some testing with WANDERER and CMYK option with 50000 lines (by the way, which is the limit?)

First test didn't quite get it 
Original1
[image]
Produced-->>
[image]

Original2:
[image]

Produced---->
[image]

The first image didn't show a lot of cyan. The second one was more balanced and imitated pretty well the colors of the original.

i-make-robots, 2018-06-09:
Wander CMYK is pretty new and pretty terrible.  Please keep issues separate so they can be dealt with correctly.

i-make-robots, 2018-06-09:
created issue #237 for CMYK Wander.

closed 2018-06-09 by i-make-robots