MarginallyClever/Makelangelo-software #252: Problems starting Makelangelo in Linux with OpenJdk 11

Navegator1, 2019-06-28:
Hello I'm trying to run Makelangelo 7.20.1 on an Ubuntu Linux with Openjdk11, and when executing the .jar  I recieve this error message:

ian@Sanguijuela:~/leechz/Librerias_Arduino/Makelangelo$ java -jar Makelangelo-7.20.1-with-dependencies.jar 
libpng warning: iCCP: known incorrect sRGB profile
java.io.tmpdir=/tmp/
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 1/1, reusable (open, marked uncloseable): 0, pending (open in creation order): 1)
X11Util: Open X11 Display Connections: 1
X11Util: Open[0]: NamedX11Display[:0, 0x7fee884b7080, refCount 1, unCloseable false]
ian@Sanguijuela:~/leechz/Librerias_Arduino/Makelangelo$ java -jar Makelangelo-7.20.1-with-dependencies.jar 
libpng warning: iCCP: known incorrect sRGB profile
java.io.tmpdir=/tmp/
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.jogamp.common.os.NativeLibrary$3 (file:/home/ian/leechz/Librerias_Arduino/Makelangelo/Makelangelo-7.20.1-with-dependencies.jar) to method java.lang.ClassLoader.findLibrary(java.lang.String)
WARNING: Please consider reporting this to the maintainers of com.jogamp.common.os.NativeLibrary$3
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Inconsistency detected by ld.so: dl-lookup.c: 111: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!

The java version shows this:
ian@Sanguijuela:~/leechz/Librerias_Arduino/Makelangelo$ java -version
openjdk version "11.0.3" 2019-04-16
OpenJDK Runtime Environment (build 11.0.3+7-Ubuntu-1ubuntu218.04.1)
OpenJDK 64-Bit Server VM (build 11.0.3+7-Ubuntu-1ubuntu218.04.1, mixed mode, sharing)

What am I doing wrong?

Thanks in advance

eljuligallego, 2020-05-04:
I have similar problem:

libpng warning: iCCP: known incorrect sRGB profile
java.io.tmpdir=/tmp/
May 04, 2020 8:50:59 PM java.util.prefs.FileSystemPreferences$1 run
INFO: Created user preferences directory.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.jogamp.common.os.NativeLibrary$3 (file:/home/julian/Makelangelo-7.21.0-with-dependencies.jar) to method java.lang.ClassLoader.findLibrary(java.lang.String)
WARNING: Please consider reporting this to the maintainers of com.jogamp.common.os.NativeLibrary$3
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Inconsistency detected by ld.so: dl-lookup.c: 111: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!

any solution?.

i-make-robots, 2020-05-04:
http://man7.org/linux/man-pages/man8/ld.so.8.html
the linker is looking a library of some kind and fails to find the right matching version... i think?

i-make-robots, 2020-05-04:
"Solved it, it really was a problem with some java update. I purged OpenJDK 11 following these instructions (https://novicestuffs.wordpress.com/2017/04/25/how-to-uninstall-java-from-linux/) and installed OpenJDK 8. It seems to be working fine now."

-- https://unix.stackexchange.com/questions/532054/program-crashes-with-the-error-inconsistency-detected-by-ld-so-dl-lookup-c-11

i-make-robots, 2020-05-05:
Did that help?

eljuligallego, 2020-05-05:
yes Dan, thanks so much for your help.

closed 2020-05-05 by i-make-robots