MarginallyClever/Makelangelo-software #253: LCD dosen't work with ramps 1.4

omarbentaleb, 2019-08-05:
i've just uploaded the makelangelo firmware to the ramps 1.4,  the servo and the stepper mptors works but lcd dosen't work all i get is a blue screen i have a reprap 12864 lcd

i-make-robots, 2019-08-05:
you can specify 128x64 type lcd. I’m travelling do I can’t look it up. LCD.h? Configure.h?

omarbentaleb, 2019-08-06:
i already did that, still dosen't work

i-make-robots, 2019-08-06:
in configure.h

    #define LCD_IS_128X64  // reprapdiscount Full Graphic Smart LCD
Controller
    //#define LCD_IS_SMART  // reprapdiscount Smart LCD Controller
(including XXL model)

should disable one and enable the other.

omarbentaleb, 2019-08-06:
yes i did that and still no luck is anything else i can do ?
 // only uncomment one of these options
#define LCD_IS_128X64  // reprapdiscount Full Graphic Smart LCD Controller
//#define LCD_IS_SMART  // reprapdiscount Smart LCD Controller (including XXL model)

i-make-robots, 2019-08-06:
We can't find our 128x64 to test results.  You'll have to experiment and
report back.  Probably something to do with refreshing the display.

omarbentaleb, 2019-08-07:
maybe i'll try other version of the firmware ?

omarbentaleb, 2019-08-07:
i also have this problem that when connect the machine to the makelangelo softwre i can controll the machine and the pen but when i open a picture in dxf or svg and i press o start nothing happens

i only have this two problem lcd all blue and when i click on start nothing happens and thank you for your help

i-make-robots, 2019-08-07:
When you press start the LCD should say "ready pen black color and click to
continue."
You cannot see the message so you don't click and it never starts.
The "click to continue" could be a separate option in config in a future
update.

omarbentaleb, 2019-08-07:
yes i've tried it now when i click start on mikelangelo software then i click on the lcd button the machine start working, but is there anyway else that i can upload the file to the sd card then plug it to the lcd and start woking  ( like a 3d printer) because the lcd is still all blue no text nothing

omarbentaleb, 2019-08-07:
here's the code in mikeangelo 

N48 G01 X-17.281 Y-22.219;*25
N49 G01 X-17.281 Y-17.281;*31
N50 G01 X-17.281 Y-12.344;*26
N51 G01 X-12.344 Y-12.344;*22
N52 G01 X-12.344 Y-17.281;*24
N53 G01 X-7.406 Y-17.281;*44
N54 G01 X-2.469 Y-17.281;*39
N55 G01 X-2.469 Y-12.344;*43
N56 G01 X-7.406 Y-12.344;*36
N57 G01 X-7.406 Y-7.406;*16
N58 G01 X-2.469 Y-7.406;*19
N59 G01 X-2.469 Y-2.469;*30
N60 G01 X-7.406 Y-2.469;*24
N61 G01 X-12.344 Y-2.469;*44
N62 G01 X-12.344 Y-7.406;*35
N63 G01 X-17.281 Y-7.406;*47
N64 G01 X-17.281 Y-2.469;*36
N65 G01 X-22.219 Y-2.469;*34
N66 G01 X-27.156 Y-2.469;*44
N67 G01 X-27.156 Y-7.406;*33
N68 G01 X-22.219 Y-7.406;*35
N69 G01 X-22.219 Y-12.344;*23
N70 G01 X-22.219 Y-17.281;*18
N71 G01 X-27.156 Y-17.281;*30
N72 G01 X-27.156 Y-12.344;*16
N73 G01 X-32.094 Y-12.344;*26
N74 G01 X-32.094 Y-17.281;*16
N75 G01 X-37.031 Y-17.281;*27
N76 G01 X-37.031 Y-12.344;*21
N77 G01 X-37.031 Y-7.406;*33
N78 G01 X-32.094 Y-7.406;*36
N79 G01 X-32.094 Y-2.469;*41
N80 G01 X-37.031 Y-2.469;*37
N81 G01 X-37.031 Y2.469;*9
N82 G01 X-37.031 Y7.406;*6
N83 G01 X-32.094 Y7.406;*13
N84 G01 X-32.094 Y2.469;*6
N85 G01 X-27.156 Y2.469;*12
N86 G01 X-22.219 Y2.469;*2
N87 G01 X-22.219 Y7.406;*15
N88 G01 X-27.156 Y7.406;*13
N89 G01 X-27.156 Y12.344;*57
N90 G01 X-22.219 Y12.344;*60
N91 G01 X-22.219 Y17.281;*48
N92 G01 X-27.156 Y17.281;*62
N93 G01 X-32.094 Y17.281;*52
N94 G01 X-32.094 Y12.344;*62
N95 G01 X-37.031 Y12.344;*53
N96 G01 X-37.031 Y17.281;*59
N97 G01 X-37.031 Y22.219;*61
N98 G01 X-32.094 Y22.219;*56
N99 G01 X-32.094 Y27.156;*52
N100 G01 X-37.031 Y27.156;*15
N101 G01 X-37.031 Y32.094;*5
N102 G01 X-37.031 Y37.031;*12
N103 G01 X-32.094 Y37.031;*7
N104 G01 X-32.094 Y32.094;*10
N105 G01 X-27.156 Y32.094;*0
N106 G01 X-27.156 Y37.031;*9
N107 G01 X-22.219 Y37.031;*5
N108 G01 X-22.219 Y32.094;*0
N109 G01 X-22.219 Y27.156;*10
N110 G01 X-27.156 Y27.156;*15
N111 G01 X-27.156 Y22.219;*3
N112 G01 X-22.219 Y22.219;*13
N113 G01 X-17.281 Y22.219;*11
N114 G01 X-12.344 Y22.219;*1
N115 G01 X-12.344 Y27.156;*13
N116 G01 X-17.281 Y27.156;*3
N117 G01 X-17.281 Y32.094;*9
N118 G01 X-17.281 Y37.031;*12
N119 G01 X-12.344 Y37.031;*0
N120 G01 X-12.344 Y32.094;*0
N121 G01 X-7.406 Y32.094;*52
N122 G01 X-7.406 Y37.031;*61
N123 G01 X-2.469 Y37.031;*48
N124 G01 X-2.469 Y32.094;*61
N125 G01 X-2.469 Y27.156;*55
N126 G01 X-7.406 Y27.156;*56
N127 G01 X-7.406 Y22.219;*52
N128 G01 X-2.469 Y22.219;*55
N129 G01 X-2.469 Y17.281;*49
N130 G01 X-2.469 Y12.344;*52
N131 G01 X-7.406 Y12.344;*57
N132 G01 X-7.406 Y17.281;*55
N133 G01 X-12.344 Y17.281;*3
N134 G01 X-17.281 Y17.281;*9
N135 G01 X-17.281 Y12.344;*5
N136 G01 X-12.344 Y12.344;*11
N137 G01 X-12.344 Y7.406;*63
N138 G01 X-17.281 Y7.406;*61
N139 G01 X-17.281 Y2.469;*48
N140 G01 X-12.344 Y2.469;*51
N141 G01 X-7.406 Y2.469;*7
N142 G01 X-7.406 Y7.406;*8
N143 G01 X-2.469 Y7.406;*5
N144 G01 X-2.469 Y2.469;*14
N145 G01 X2.469 Y2.469;*34
N146 G01 X2.469 Y7.406;*45
N147 G01 X7.406 Y7.406;*32
N148 G01 X7.406 Y2.469;*35
N149 G01 X12.344 Y2.469;*23
N150 G01 X17.281 Y2.469;*18
N151 G01 X17.281 Y7.406;*31
N152 G01 X12.344 Y7.406;*17
N153 G01 X12.344 Y12.344;*37
N154 G01 X17.281 Y12.344;*47
N155 G01 X17.281 Y17.281;*35
N156 G01 X12.344 Y17.281;*45
N157 G01 X7.406 Y17.281;*25
N158 G01 X7.406 Y12.344;*27
N159 G01 X2.469 Y12.344;*22
N160 G01 X2.469 Y17.281;*17
N161 G01 X2.469 Y22.219;*23
N162 G01 X7.406 Y22.219;*24
G00 F50 Z90
G00 F500 
G00 F50 Z90
G00 F500 
Generator=Outline paper
15 line segments. 41.189266cm Estimated draw time: 54s .
Completed

HELLO WORLD! I AM POLARGRAPH #-1
Firmware v9
D10 
== http://www.marginallyclever.com/ ==
M100 - display this message
M101 [Tx.xx] [Bx.xx] [Rx.xx] [Lx.xx]
- display/update board dimensions.
As well as the following G-codes (http://en.wikipedia.org/wiki/G-code):
G00,G01,G02,G03,G04,G28,G90,G91,G92,M18,M114

HELLO WORLD! I AM POLARGRAPH #-1
Firmware v9
== http://www.marginallyclever.com/ ==
M100 - display this message
M101 [Tx.xx] [Bx.xx] [Rx.xx] [Lx.xx]
- display/update board dimensions.
As well as the following G-codes (http://en.wikipedia.org/wiki/G-code):
G00,G01,G02,G03,G04,G28,G90,G91,G92,M18,M114

> D10 V5
M101 A0 T325 B-325 
M101 A1 T500 B-500 
M101 A2 T170 B10 
G92 X0.0 Y283.0
D6 X0 Y283
G0 F500 A300 
M17

> Saving dimensions.
(-325.00,0.00,10.00) - (325.00,250.00,33.00)

> Saving dimensions.
(-325.00,-500.00,10.00) - (325.00,500.00,33.00)

> Saving dimensions.
(-325.00,-500.00,10.00) - (325.00,500.00,170.00)

> Saving home.

G91
G00 X-100 Y0
G90
G00 X0 Y283
M110 N0
N0 G90;*43
N1 M06 T0;*107
N2 G00 F500 A300;*33
N3 M117 Change to black Click to continue;*18
N4 M300 S60 P250;*45
N5 M226;*27
> pausing ended.

> SD is added
N6 M117;*25

> SD is removed
N7 G00 F50 Z90;*53

N8 G00 F500;*121
N9 M06 T0;*99
N10 G00 F500 A300;*18
N11 M117 Change to black Click to continue;*33
N12 M300 S60 P250;*26
N13 M226;*44
> pausing ended.

> SD is added
N14 M117;*42

> SD is removed
N15 G00 X-39.5 Y39.5;*58

N16 G00 X-39.5 Y39.5;*57
N17 G01 F50 Z160;*59
N18 G01 F500;*73
N19 G01 X39.5 Y39.5;*26
N20 G01 X39.5 Y-39.5;*61
N21 G01 X-39.5 Y-39.5;*17
N22 G01 X-39.5 Y39.5;*63
N23 G00 F50 Z90;*3
N24 G00 F500;*71
N25 G00 X0 Y33;*36
G00 F50 Z90
G00 F500

omarbentaleb, 2019-08-07:
could it be a hardware problem because i've changed the lcd and still the stame maybe it's from the board but the lcd lights on

i-make-robots, 2019-08-07:
A bad way to do it would be to start immediately as soon as an SD card is
put in that has just one file.

A better way is to find out why nothing shows up on your LCD.

omarbentaleb, 2019-08-07:
what do you mean by start immediately ? and the file file does it need to be an .ngc

omarbentaleb, 2019-08-07:
how can i do that when i plug the sd card with a .ngc file what do need to do ?

omarbentaleb, 2019-08-07:
do you know how to fix this ?

i-make-robots, 2019-08-07:
There is no easy fix.  If you have a 20x4 lcd then the problem would just be “figure out the right pins” or “prove the lcd works”. AFAIK you have only a 128x64 panel so I can’t test.  I’m fully booked right now.  I can get one if you’re willing to pay for my time + parts.

omarbentaleb, 2019-08-08:
yes i do have a 20x4 lcd but when i plug it in instead of the 128x64 it only lights up, is there ny way else i can make it work

i-make-robots, 2019-08-08:
Ok, then it sounds like the pins to make the lcd work from ramps are wrong. I have both ramps and the lcd so I can try here.

I don’t appreciate the way you’re asking for help here and also posting to places that are not relevant, which then automatically emails me.  It’s making extra noise in my life.  I’ll try your setup tomorrow morning and get back to you with results.

omarbentaleb, 2019-08-08:
i really appreciate your help and i mean no disrespect i only posted in other places only to get more help and sorry for my bad english i'm really trying my best, and i can't wait for your results 
thank you so much

i-make-robots, 2019-08-09:
Ok, I tested makelangelo-firmware with ramps 1.4 and the reprapdiscount
smart controller (20x4 lcd).
The backlight is on but no text appears.

Then I tested the Arduino liquidCrystal "hello world" demo with the pin
definitions in the makelangelo-firmware.  The LCD back light does not come
on.  (?!)

Then I compared the pin definitions with Marlin.  they match.
Then I compared the pin definitions with reprapdiscount's official product
page.  They match.

So at this point I have no idea why the LCD does not light up.
I'm asking around.

omarbentaleb, 2019-08-09:
ok tell me please if you a solution, tahnk you

omarbentaleb, 2019-08-09:
i also have this weird problem sometime one or the two stepper start vibrating and making noise and th driver get really hot and sometimes it works normally , do you have any idea ho to fix this ?

i-make-robots, 2019-08-09:
No.  A "sometimes" problem is deeply suspicious.

i-make-robots, 2019-08-09:
i asked other experts for help:
https://forum.arduino.cc/index.php?topic=630686.0

omarbentaleb, 2019-08-09:
ok i really appreciate your help, but this new problem is driving me crazy the robot was working with no problem all day long sudently and shut it off and on and the right stepper start making noise and vibrating i've changed the stepper the cable, the driver and it's still the same problem and the driver is getting verry hot i've even tested it out without the weight still the same problem and this problem happended to me in the beginning then it stoped and it happened again and i even changed the board still the same problem, i think the problem is from the firmware.

omarbentaleb, 2019-08-09:
and i'm using a 12v 15a power supply, if this help

omarbentaleb, 2019-08-10:
and now it works with no problem, do you have an idea on what could be the problem

omarbentaleb, 2019-08-10:
but this dosen't matter i could fix it my main problem is the lcd and sd card

i-make-robots, 2019-08-12:
why did it suddenly start working?

omarbentaleb, 2019-08-13:
i don't know maybe it's was bad connection between the wires but now it works, and for the lcd did find any solution ? thank you

omarbentaleb, 2019-08-14:
is there a way to program it that when i out the sd card card and press the button it will start it job

i-make-robots, 2019-08-14:
I am writing something like that. I will post when it is working.

omarbentaleb, 2019-08-14:
ok thanks i really appreciate it

omarbentaleb, 2019-08-22:
Hi how are you ? hope you doing great. any news about the new program

i-make-robots, 2019-08-22:
Nope.  I've had no time to look into it.  RAMPS support is very low
priority in my world right now, I'm under a big deadline for another
project.

omarbentaleb, 2019-08-22:
Ok no problem, takeyour time. thanks

omarbentaleb, 2019-09-13:
Ok i will, thanks

omarbentaleb, 2019-11-21:
Hi men how are you ? i'm still waiting on the software as i said i really need it.

i-make-robots, 2019-11-21:
It's on my todo list, right after "adjustable home position for m3".

i-make-robots, 2019-11-21:
I don't have a mega+ramps+LCD setup to test with, I can't tell how to make this work.  We don't use that combination in our official robot.  If you get it working please send in a pull request and we'll make sure your fix is permanent.  Otherwise it's very unlikely I'll get the time in my schedule to fix this soon.

i-make-robots, 2020-06-25:
Closing this issue because it is a makelangelo-firmware bug, not an app bug.

closed 2020-06-25 by i-make-robots