MarginallyClever/Makelangelo-software #255: Custom parameters use ridiculous belt lengths...

akzmonster, 2019-09-04:
I am using a MEGA/Ramps 1.4 with two 4988/Nema 17s (0.9) set to 1/16 microstepping.

The width is 955 mm between motor centers and 533 to the bottom of the board.

If the machine were to be able to draw to the extreme bottom corners of the board the belts would need to be at most 1094 mm each. But the custom settings show that I need 2 m belts each. This is crazy. But I decided to just so what it says and not deviate.

When I select "Find Home" it runs to the end of each 2 m belt and then sets the home (0,0) to be about a meter under the board. 

Lol. Is this just not made for custom setups?

i-make-robots, 2019-09-04:
Yes.  The default m3 is 2m on each belt.  The m5 is 1m on each belt.  You can set belt length with D6 Lnnn Rnnn.  If you start the machine at the home position you can ask it to measure the belt for you with D12 (recordHome).

closed 2019-09-04 by i-make-robots

akzmonster, 2019-09-04:
Perfect thanks. Is there a manual for this anywhere? Maybe I'm missing
something.

i-make-robots, 2019-09-04:
The variety of DIY options is too damn high.  For our supported models, we
have instructions here: https://mcr.dozuki.com/

akzmonster, 2019-09-04:
Thanks. But do you list the DIY options anywhere? I looked through the instructions but I couldn't seem to find any codes other than D11.

i-make-robots, 2019-09-04:
Once you've used the firmware to find your belt length and set your motor
step size (look in configure.h) then the rest should be do-able in the
software - set machine size, set pen up/down, set max speed & acceleration,
set pen size.
That should probably be everything, unless you have a really odd clock
speed or something exotic.

Maybe if I knew better what you were trying to change I could be more
specific.

akzmonster, 2019-09-04:
Oh oh oh got it! I'm going back to firmware setup for this one. Thanks for
the help! Promise I'm done crowing about it. 😉

i-make-robots, 2019-09-04:
Can you edit the wiki?  maybe add notes about what you've learned to help
anyone coming after you.

i-make-robots, 2019-09-04:
I'm kind of excited that you're playing in the dev branch.  I just added
"adjust acceleration based on where the pen is" so that the machine runs
faster at the top and slower at the bottom.  Should improve image
fidelity.  Nobody else afaik has tried it and I'm eager to hear your
thoughts.