MarginallyClever/Makelangelo-software #257: Some of the converters draw a line from 0,0

mishafarms, 2019-12-03:
Dan,
  While playing with the Makelangelo software. I came across some times when convert an image that there is a line drawn from 0,0 to the beginning of the drawing. Spiral pulse was one.

I have fixes for these. If you want them.

Michael (github@mishafarms)

i-make-robots, 2019-12-03:
Yes please!

i-make-robots, 2020-01-31:
which converters have the issue, please?

mishafarms, 2020-01-31:
I thought I gave you the code already. One of the convertors was Pulsing Spiral and the other was SandyNoble.

If you look at my repository in the pi4 branch you should see all the fixes there. 

My repository isn't quite up to date with yours at the moment. I will merge in a few days. I am a little sick and playing with something else at the moment.

Michael

closed 2020-01-31 by mishafarms