MarginallyClever/Makelangelo-software #258: Doesn't run on Raspberry Pi 4

mishafarms, 2019-12-03:
Dan,

    I am trying to run the Makelangelo software on a Raspberry Pi 4, The issue is the jogamp library from Maven. It is not the version that will run on a Raspberry Pi 4, it should run on a Pi 3 (I have not yet tried it). The version they recommend is jogamp-fat.

I can get this running in my branch, I just downloaded the library and use it. But I am not sure if you would want to support Raspberry Pi 4 or not and if so I am not sure how to make the change to use the local library in maven.

I am willing to do the work, if you will guide me.

Also I have the Raspberry Pi taking camera captures from the Makelangelo software as well if you are interested in that.

I also have a current pull request for a fix to Scratch2 file import.
You can email me at github@mishafarms.us

i-make-robots, 2019-12-03:
I am delighted to help you.  If you want to join me on discord for live
chat, https://discord.gg/KNgbqr

i-make-robots, 2019-12-03:
Scratch 2?  AFAIK only version 3 is available now.  How do you make Scratch 2 programs?

i-make-robots, 2019-12-03:
https://scratch.mit.edu/download/scratch2  omfg.  yes!

i-make-robots, 2019-12-03:
I suppose you'd be interested in separating the GUI from the application, so there could be a CLI version of makelangelo-software?

i-make-robots, 2020-01-17:
> I am not sure if you would want to support Raspberry Pi 4 or not
Yeah, all pis would be neat.

> I am willing to do the work, if you will guide me.
Help me help you :)

mishafarms, 2020-01-17:
Dan,

     For now I am willing to support the Raspberry Pi. 

Michael